gqrx-sdr/gqrx #585: RSP2 with gnuradio

Mitchell1314, 2018-01-12:
Hi, I am using sdrplay rsp2 recently. And I am trying to make rsp2 work with gnuradio. Here is my  grc:
[image]

when I run it, I got nothing:
    
[image]

 Can you give me some help about this? Waiting for your reply. Thanks.

willcode, 2018-01-12:
> default

Yes, I'm just about finished with better support for the SDRplay devices 
in gr-osmosdr. You can try it out without overwriting existing software 
on your system. Please let me know how it works for you.

First, you have to install the latest SDRplay SDK (v2.11 15NOV2017). See 
their site. Then:

git clone https://github.com/willcode/gr-osmosdr
cd gr-osmosdr
git checkout sdrplay2
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/tmp/test -DENABLE_NONFREE=yes ..

(make sure SDRplay shows up in the enabled list)

mkdir /tmp/test
make install
LD_LIBRARY_PATH=/tmp/test/lib64:$LD_LIBRARY_PATH gqrx

willcode, 2018-01-12:
Note that the osmocom source in GNU Radio has a generic view of gains, 
and they don't map to the way RSPs work (attenuation from max gain). 
You'll end up with default (max) LNA gain and controllable baseband 
attenuation (20-59 dB) in the gain field. Confusing right now, but can 
be cleaned up with a new GRC XML file (GRC's view of a block) for SDRplay.

csete, 2018-01-12:
Thanks for taking this @willcode!

@Mitchell1314, I feel compelled to point out, that this is a bug tracker for the Gqrx project, not a tech support forum for GNU Radio. I will let you finish the conversation in peace, but please use the GNU Radio forums for such matters in the future.

Mitchell1314, 2018-01-13:
I feel so sorry about this. Thanks for your remind

closed 2018-01-28 by csete