gqrx-sdr/gqrx #589: Gain control / AGC for AirspyHF+

streamofstars, 2018-01-28:
I am using version 2.10 (on Linux), checking/unchecking the AGC checkbox has zero effect, sliders for manual gain control also not available. Are these by design? I am getting much better reception with SDRSharp on Windows in the same location, on the same antenna.

csete, 2018-01-28:
Airspy HF+ has no gain controls, but you already know that if you have used SDR#.
User support is provided through mailing list: http://gqrx.dk/user-support

closed 2018-01-28 by csete

streamofstars, 2018-01-28:
Be advised that checking/unchecking AGC box in SDRSharp actually makes audio signal lauder/quieter (I have no idea on mechanics of this, since AGC in AirspyHF+ is controlled internally).

If technically possible, please consider making hardware-locked features also unavailable software-side e.g. locked, greyed-out AGC checkbox in this case. I would have avoided the confusion. Thank you.

csete, 2018-01-28:
I think you are confusing hardware AGC with software AGC. The Airspy HF+ has no programming interface for setting any gain.

The AGC control you are referring to in SDR# must be the software AGC. Gqrx has that too under "Receiver Options". The hardware AGC is only available for some devices like RTLSDR. Unfortunately, gqrx can not know whether a device supports hardware AGC or not, so the only options I have is to show it or not for all devices.