gqrx-sdr/gqrx #590: Nuand Bladerf xb200 transverter

j4mbob, 2018-01-31:
lastest gqrx compiled from github source (2.10)
lastest gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.11
latest libbladerf compiled from github
Fedora 25

Initialising bladerf within gqrx with xb200 transverted enabled does not allow frequency range down to the transverter range. gqrx wont go below 237mhz.

This is being intialised via soapy with the following string: soapy=0,driver=bladerf,xb200=custom

I can see in the console output that gqrx is correctly passing this to soapy:

[jamie@jamie-desktop buil]$ gqrx 
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.11
built-in source types: file fcd rtl_tcp hackrf bladerf rfspace soapy redpitaya 
Resampling audio 96000 -> 48000
BookmarksFile is /home/jamie/.config/gqrx/bookmarks.csv
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.11
built-in source types: file fcd rtl_tcp hackrf bladerf rfspace soapy redpitaya 
[INFO] bladerf_open_with_devinfo()
[INFO] bladerf_get_serial() = d8de7494d13a60ee961497ac6004ecdf
[INFO] setSampleRate(1, 1.000000 MHz), actual = 1.000000 MHz
[INFO] setSampleRate(0, 1.000000 MHz), actual = 1.000000 MHz
[INFO] bladeRF: XB200 is attached
[INFO] bladeRF: Set XB200 filterbank 'custom'
[INFO] bladeRF: Adjusting mix path through XB200
[INFO] setSampleRate(1, 1.000000 MHz), actual = 1.000000 MHz
[INFO] setSampleRate(1, 50.000000 MHz), actual = 50.000000 MHz

But gqrx doesnt appear to take any notice of it.

Interestingly, if i call bladerf in gqrx via gr-osmosdr directly and bypass Soapy with string bladerf=0,xb200=custom in gqrx, then gqrx does correctly detect that im using the transverter board.

Any suggestions?

csete, 2018-01-31:
Sorry, but I don't think this is a bug in gqrx.

Gqrx requests the frequency range from gr-osmosdr and has otherwise no idea about the hardware configuration. So I can't make "gqrx take notice of the xb200".
Gqrx does not talk directly to SoapySDR either; SoapySDR is just a backend in gr-osmosdr. If the frequency range is wrong when using soapysdr, then the problem is somewhere between SoapySDR and gr-osmosdr.

closed 2018-01-31 by csete

j4mbob, 2018-01-31:
Apologies -  when using it via Soapy you just need to tick "no limits" in GQRX and that allows you to get the full frequency range!

Works fine now.

csete, 2018-01-31:
I thought about mentioning that because disabling limits usually works in these circumstances; however, I  still think something is broken in the SoapySDR / gr-osmosdr integration since it doesn't report the correct frequency range.