gqrx-sdr/gqrx #593: HackRF not more supported?

andimik, 2018-02-03:
I have an older version from April 2017, which worked with my HackRF.

>  getDeviceList : Available input devices:
>     0 : "HackRF HackRF One 46645f"
>     1 : "RFSPACE SDR-IQ Receiver"
>     2 : "RFSPACE SDR-IP Receiver"
>     3 : "RFSPACE NetSDR Receiver"
>     4 : "RFSPACE Cloud-IQ Receiver"
>     5 : "RTL-SDR Spectrum Server"
>     6 : "Red Pitaya Transceiver Server"
>     7 : "Complex Sampled (IQ) File"

[image]

Today I compiled the latest from source, but I get

> FATAL: No supported devices found to pick from.
> 
> Trying to fill up 1 missing channel(s) with null source(s).
> This is being done to prevent the application from crashing
> due to gnuradio bug #528.
> 
> 
> Failed to set RX input rate to 1e+07
> Your device may not be working properly.
> 
> Decimation: 4
>   stage: 1  ratio: 4

[image]

csete, 2018-02-03:
Oh come on, give me a break!

closed 2018-02-03 by csete

csete, 2018-02-03:
But to answer your question, hackrf is still supported, so your build must be broken.

andimik, 2018-02-04:
I tried 2.7 and 2.9. There are no warnings or error in the `cmake` or `make` output. But: Both did not show HackRF in the I/Q input window. 

See attached file: 
[gqrx_2.9_build_output.txt](https://github.com/csete/gqrx/files/1692587/gqrx_2.9_build_output.txt)

Do I have to specify something in `cmake` that I have a HackRF?

shrishailayya, 2018-02-04:
Dear All 
I had installed GNU radio previously using following procedure 
https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux 

but in that GNU-rado I didn't find rtl-sdr lib so tried following command 
git clone git://git.osmocom.org/gr-osmosdr
cd gr-osmosdr/
mkdir build
cd build/
cmake ../
make
sudo make install
sudo ldconfig 

but I got two log files which I have attached. Please if anybody can help me installing osmocom RTL SDR it would very helpful thank you. 

Please find command history: 

shrish@shrish-OptiPlex-990:~$ git clone git://git.osmocom.org/rtl-sdr.git 
Cloning into 'rtl-sdr'... 
fatal: unable to connect to git.osmocom.org: 
git.osmocom.org[0: 144.76.43.76]: errno=Connection timed out 
git.osmocom.org[1: 2a01:4f8:191:444b::2:5]: errno=Network is unreachable 

shrish@shrish-OptiPlex-990:~$ git clone http://cgit.osmocom.org/cgit/rtl-sdr/ 
Cloning into 'rtl-sdr'... 
Checking connectivity... done. 
shrish@shrish-OptiPlex-990:~$ git clone http://git.osmocom.org/rtl-sdr.git 
fatal: destination path 'rtl-sdr' already exists and is not an empty directory. 
shrish@shrish-OptiPlex-990:~$ cd gr-osmosdr/ 
shrish@shrish-OptiPlex-990:~/gr-osmosdr$ mkdir build 
mkdir: cannot create directory ‘build’: File exists 
shrish@shrish-OptiPlex-990:~/gr-osmosdr$ cd build/ 
shrish@shrish-OptiPlex-990:~/gr-osmosdr/build$ cmake ../ 
-- Build type not specified: defaulting to release. 
-- Extracting version information from git describe... 
-- Configuring Boost C++ Libraries... 
-- Boost version: 1.58.0 
-- Found the following Boost libraries: 
-- thread 
-- system 
-- chrono 
-- date_time 
-- atomic 
-- Checking for module 'gruel' 
-- No package 'gruel' found 
-- Could NOT find GRUEL (missing: GRUEL_LIBRARIES GRUEL_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-core' 
-- No package 'gnuradio-core' found 
-- Could NOT find GNURADIO_CORE (missing: GNURADIO_CORE_LIBRARIES GNURADIO_CORE_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-iqbalance' 
-- No package 'gnuradio-iqbalance' found 
-- Could NOT find GNURADIO_IQBALANCE (missing: GNURADIO_IQBALANCE_LIBRARIES GNURADIO_IQBALANCE_INCLUDE_DIRS) 
-- gnuradio-uhd not found. 
-- Could NOT find GNURADIO_UHD (missing: GNURADIO_UHD_INCLUDE_DIRS) 
-- gnuradio-fcd not found. 
-- Could NOT find GNURADIO_FCD (missing: GNURADIO_FCD_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-fcdproplus' 
-- No package 'gnuradio-fcdproplus' found 
-- gnuradio-fcdproplus not found. 
-- Could NOT find GNURADIO_FCDPP (missing: GNURADIO_FCDPP_LIBRARIES GNURADIO_FCDPP_INCLUDE_DIRS) 
-- Checking for module 'libosmosdr' 
-- No package 'libosmosdr' found 
-- libosmosdr not found. 
-- Checking for module 'librtlsdr' 
-- No package 'librtlsdr' found 
-- librtlsdr not found. 
-- Checking for module 'libmirisdr' 
-- No package 'libmirisdr' found 
-- libmirisdr not found. 
[... 63 lines cut ...]

-- Configuring incomplete, errors occurred! 
See also "/home/shrish/gr-osmosdr/build/CMakeFiles/CMakeOutput.log". 
See also "/home/shrish/gr-osmosdr/build/CMakeFiles/CMakeError.log". 
shrish@shrish-OptiPlex-990:~/gr-osmosdr/build$ make 
make: *** No targets specified and no makefile found. Stop. 
shrish@shrish-OptiPlex-990:~/gr-osmosdr/build$ sudo make install 
[sudo] password for shrish: 
make: *** No rule to make target 'install'. Stop. 
shrish@shrish-OptiPlex-990:~/gr-osmosdr/build$ sudo ldconfig 
shrish@shrish-OptiPlex-990:~/gr-osmosdr/build$ gnuradio-companion <<< Welcome to GNU Radio Companion 3.7.10.1 >>> 

Block paths: 
/usr/local/share/gnuradio/grc/blocks 
shrish@shrish-OptiPlex-990:~/gr-osmosdr/build 

With Best Regards 
Shrishail Hiremath, Assistant Professor 
Department of Electronics and Communication Engineering 
National Institute of Technology 
Rourkela, PIN - 769008 
ORISSA, INDIA 
hiremaths@nitrkl.ac.in 
Mob:9438503621 
Office:0661-2462459

csete, 2018-02-04:
Sorry @andimik but it's not my job to teach people to build software from source and certainly not when they are not even familiar with the information available in the readme and the website.

@shrishailayya This is a bug tracker, not a tech support forum.

csete, 2018-02-04:
Gqrx is device-agnostic. Interfacing to the hardware is done by the gr-osmosdr library. When you build gr-osmosdr it will look for driver libraries and header files and enable support according to availability. A summary of device support is printed at the end of gr-osmosdr cmake.

All this makes me wonder, how the hell you could have a working gqrx before. Sounds like you have a nice mixture of packages installed from binary packages and sources, in which case I can only wish you the best of luck sorting it out.

Anyway, this is not a bug in gqrx and the discussion does not belong here, see http://gqrx.dk/user-support . While you are on the website, please take a look at the download page and see whether there are other installation options that might be more suitable for you than installing from source.