gqrx-sdr/gqrx #598: Request - CTCSS tone filtering, decoding

erikarn, 2018-02-18:
hi,

Listening to repeaters using GQRX is difficult because in a lot of instances the PL tone is very loud compared to the rest of the audio signal.

Would someone please consider linking in the GNU radio block which filters out the CTCSS signals, and optionally decode it and show the tone frequency? I know there's not a lot of space in the UI but I think this will drastically help the usefulness of GQRX on repeaters.

Thanks!

-a

csete, 2018-02-18:
Already exists as issue #362.

csete, 2018-02-18:
I thought about adding a simple audio filter to the FM demodulator, which would solve this particular problem. Unfortunately, whatever little time I have for gqrx I have been spending on user support.

erikarn, 2018-02-18:
hi,

Would a donation help? I know what it's like to get unsolicited requests
from users on an open source project. :)

Or, if you point out where I should be looking in the gqrx source, I can
give it a crack. I wondered about extending the selector widget (the
bar/centre markers) to allow it to be left right, rather than centre/width,
and then putting the filter over /that/...

-a

willcode, 2018-02-18:
It looks like you would connect in a high-pass filter in 
dsp/rx_demo_fm.cpp right after d_quad. It shouldn't matter whether it's 
before or after d_deemph. There is already a low-pass filter (dsp/lpf.h, 
dsp/lpf.cpp) that you could use as an example.

That should be enough to experiment with a fixed filter. Making it 
controllable by the GUI is another thing.

willcode, 2018-02-18:
Ignore that! The low-level filtering functionality is already there, so 
you're just looking for how to do the GUI work and squelch?

erikarn, 2018-02-18:
Ctcss decode and squelch activation can come later.

Right now theres no way to filter out audio frequency ranges. The obvious
thing is to put in a high pass filter above 200Hz to get rid of ctcss tones
but it would also be nice to do audio post processing of wider decode
bandwidth.

So e.g. putting in the gui hooks to pass upper and lower ranges into a bpf
would be nice.

Ctcss squelch activation would be nice but that can come later. :)

A

But, rx_filter works on complex, so you'd have to dup that and make a
float version to work with audio.

> Ignore that! The low-level filtering functionality is already there, so
> you're just looking for how to do the GUI work and squelch?
>
>> It looks like you would connect in a high-pass filter in
>> dsp/rx_demo_fm.cpp right after d_quad. It shouldn't matter whether
>> it's before or after d_deemph. There is already a low-pass filter
>> (dsp/lpf.h, dsp/lpf.cpp) that you could use as an example.
>>
>> That should be enough to experiment with a fixed filter. Making it
>> controllable by the GUI is another thing.
>>
>>> hi,
>>>
>>> Would a donation help? I know what it's like to get unsolicited requests
>>> from users on an open source project. :)
>>>
>>> Or, if you point out where I should be looking in the gqrx source, I can
>>> give it a crack. I wondered about extending the selector widget (the
>>> bar/centre markers) to allow it to be left right, rather than
>>> centre/width,
>>> and then putting the filter over /that/...
>>>
>>>
>>> -a
>>>
>>>
>>>
>>>  > I thought about adding a simple audio filter to the FM
>>> demodulator, which
>>>  > would solve this particular problem. Unfortunately, whatever
>>> little time I
>>>  > have for gqrx I have been spending on user support.
>>>  >
>>> or mute
>>>  > the thread
IYVmNYXVijgjY919nks5tV-8rgaJpZM4SJmQm>
>>>
>>>  >
>>>
>>> <https://github.com/csete/gqrx/issues/598#issuecomment-366535330>, or
>>> mute the thread
ABMzNzrgcG3WsUzniTFklNruRdyied2Aks5tWGiwgaJpZM4SJmQm>.

—
<https://github.com/csete/gqrx/issues/598#issuecomment-366539924>, or mute
the thread

csete, 2018-02-18:
Ok, I guess this is the "opposite" of issue #362.

A simple band pass filter after the FM demodulator would do the trick. I am working on another SDR project at the moment because I need TX and want to get rid of the dependencies, but feel free to experiment.

kantooon, 2018-02-19:
I can add the CTCSS decode support. My main problem with that is the UI. The rest is pretty simple. Let me see it I can find some spare time to work on it.

stillline, 2019-06-11:
CTCSS / DCS identification is only possible with SDR# and a plugin. I've recently switched to a mac so SDR# is no longer an option for me.  

I know that development is time consuming and you have more important work to do but this feature would be greatly appreciated. 

I use GQRX daily to find and identify frequencies in very busy urban environments and knowing which codes a station is transmitting is very helpful in identifying two businesses using the same frequency with different tones.

craigerl, 2021-06-29:
sox can filter out the pl tones,  run gqrx, press the [UDP] button, then run

socat stdout udp-listen:7355 | play   -t raw -b 16 -e signed  -r 48k - -r 16k sinc -n 5000 180-8000

ckww, 2021-07-03:
> sox can filter out the pl tones, run gqrx, press the [UDP] button, then run
> 
> socat stdout udp-listen:7355 | play -t raw -b 16 -e signed -r 48k - -r 16k sinc -n 5000 180-8000

Craigerl, I tried your command line, but I don't have the play package installed.  Can you tell me the package name I need for Linux Mint (Ubuntu)?

I used aplay, my Alsa player, and I can listen to the streaming audio.  The concept is there.

Thank you.

ckww, 2021-07-04:
OK, in further research, I find sox is the package I needed to install to use the play command.  (It was mentioned in craigerl's post first thing!)
My question here is answered.  Thank you.

craigerl, 2021-07-07:
the ubuntu package is called "sox", it provides the play command in this situation.   The "-n" parameter makes the filter steep, but introduces a delay as a tradeoff.  Mostly undocumented as far as I can tell.. Works great tho.