gqrx-sdr/gqrx #600: Please don't sort the bookmarks

lutusp, 2018-02-25:
I have a rather long list of bookmarks, broken into categories, which I convert to a CSV file from a database for use with Gqrx, but the present default of sorting them by frequency scrambles them and makes them unusable.

My bookmarks are grouped into different categories that need to be kept separate, but because their frequencies are unrelated to their categories, the present default of sorting by frequency is the worst possible choice.

You could arrange to sort any column by clicking the header for that column, but that sounds like a lot of work (I know, I've done it many more times than I care to remember). It would be simpler to not sort at all, and that would be better IMHO.

Maybe a selectable option to sort or not sort?

Thanks, and BTW Gqrx is a great program.

csete, 2018-02-25:
Thanks for the feedback. I don't anticipate any further development of gqrx because I am working on a new SDR project, which will eventually replace gqrx, but I will take all input into consideration.

csete, 2018-03-17:
I took a look at this to see if I could make some click-to-sort, but it requires moire time than what I have at the moment, sorry.

In case you build gqrx from source, here is the line that sorts the frequencies:
https://github.com/csete/gqrx/blob/master/src/qtgui/bookmarks.h#L84

To disable sorting completely, search for std::stable_sort in
https://github.com/csete/gqrx/blob/master/src/qtgui/bookmarks.cpp

kolbma, 2020-03-11:
Hi @lutusp 

Here is a tree https://github.com/kolbma/gqrx/tree/bookmarks-ext which introduces some changes to bookmarks...

One part is **not sorted** by default for bookmarks, but manually sortable by clicking the column headers. In the context menu there is an action to reset the sorting.

There is also another **column** **"Info"** where you can put in some description for the bookmark.
If you use this column, the bookmark.csv file is **not backward compatible**.
So make a backup of your bookmark.csv before. And do **not call it bookmark.csv.bck** in the same folder, because another part is, that saving the bookmarks has changed.
The official version overwrites the existing file. If there is some problem during write (crash, disk full, whatever) your bookmarks are lost. My changeset introduces a failsafe save and uses the before called **bookmark.csv.bck** to backup the current file, before moving a temporary file in place, and deleting the backup on success.

Tags are also editable now. And the plotter reacts to changes on the fly.

I'll test the changes the next days and will make a pull request.
But it would be nice if someone could test my changes, because maybe there are use-cases of bookmarks which I don't think of.