gqrx-sdr/gqrx #601: gnuradio

kasmf, 2018-03-01:
how to uninstall gnuradio installed from source (hint installed from: https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux) in order to install new version ?

csete, 2018-03-01:
Why can't people understand that this is a gqrx bug tracker and not a freaking help desk...

closed 2018-03-01 by csete

alexf91, 2018-03-01:
It seems to get worse. Maybe an issue template with a link to the mailing list would help.
https://blog.github.com/2016-02-17-issue-and-pull-request-templates/

csete, 2018-03-01:
I thought of that, but it I doubt I am able to explain this even in a template to people who think this is the place to post gnuradio questions. We'll see, for now I can just close issues.

cbryant1000, 2018-03-01:
Maybe block repeated offenders?

Cliff Bryant                  __

___________________/_O_______________________________

 email: cbryant1000@gmail.com | cell: (617) 417-1414 | KB1YQL