gqrx-sdr/gqrx #603: Changes in gnuradio-audio

szg0000, 2018-03-03:
```
user@debian:~/gqrx.git/build$ qmake ..
Project MESSAGE: No prefix given. Using /usr/local
Project MESSAGE: Gqrx configured with gnuradio-audio backend.
Project ERROR: gnuradio-audio development package not found
```

But there is no more gnuradio-audio package in Debian.
The new package is `"libgnuradio-audio3.7.10 - gnuradio audio functions"` but does not contain source files.

alexf91, 2018-03-03:
What about gnuradio-dev?

csete, 2018-03-03:
gnuradio-audio is the component name in the gnuradio config file, not a library name.

This primary audio backend is pulseaudio. The secondary is portaudio. Only people who really hate themselves should use gnuradio-audio on linux.

If you have problems building on debian, I suggest you take a look at how the debian project builds gqrx. They always build the latest version and I get their patches if something is wrong.

closed 2018-03-03 by csete

csete, 2018-03-03:
Nothing has changed, this is with gnuradio 3.7.11:

```
$ pkg-config --list-all | grep gnuradio
gnuradio-filter                gnuradio-filter - GNU Radio's filter signal processing blocks
gnuradio-iqbalance             gnuradio-iqbalance - gnuradio I/Q balancing
gnuradio-osmosdr               gnuradio-osmosdr - GNU Radio block for various radio hardware
gnuradio-audio                 gnuradio-audio - The GNU Radio block for all supported audio sound systems
gnuradio-qtgui                 gnuradio-qtgui - GNU Radio blocks for QT GUI
gnuradio-channels              gnuradio-channels - GNU Radio's channel model blocks
gnuradio-blocks                gnuradio-blocks - GNU Radio basic block library
gnuradio-analog                gnuradio-analog - GNU Radio blocks for analog communications
gnuradio-runtime               gnuradio-runtime - GNU Radio core runtime infrastructure
gnuradio-iio                   gnuradio-iio - GNU Radio blocks for the IIO devices
gnuradio-dtv                   gnuradio-dtv - GNU Radio's digital TV signal processing blocks
gnuradio-wavelet               gnuradio-wavelet - Wavelet signal processing blocks for GNU Radio
gnuradio-fcdproplus            gnuradio-fcdproplus - GNU Radio block for Funcube Dongle Pro+
gnuradio-noaa                  gnuradio-noaa - GNU Radio blocks implementing a NOAA satellite protocols
gnuradio-atsc                  gnuradio-atsc - The GNU Radio blocks for ATSC decoding
gnuradio-vocoder               gnuradio-vocoder - GNU Radio blocks implementing voice codecs
gnuradio-fcd                   gnuradio-fcd - Funcube Dongle source block for GNU Radio
gnuradio-fec                   gnuradio-fec - GNU Radio's FEC signal processing blocks
gnuradio-trellis               gnuradio-trellis - GNU Radio blocks for trellis-based processing
gnuradio-digital               gnuradio-digital - GNU Radio blocks for digital communications
gnuradio-fft                   gnuradio-fft - GNU Radio's FFT signal processing blocks
gnuradio-pager                 gnuradio-pager - GNU Radio blocks implementing a FLEX pager decoder

```

szg0000, 2018-03-03:
`"apt-get install pkg-config"` solved the problem.
I suggest to implement an error message which describes the real missing component.

csete, 2018-03-03:
Good idea... Do you also have a suggestion how to do that?
Patch is welcome.