gqrx-sdr/gqrx #606: Better Airspy HF+ support

elibaum, 2018-03-06:
Hello csete,

I have been a long time user of gqrx and really do love it!

I was wondering if you could add support (through the gqrx gui) for some of the manual settings which have been enabled in newer Airspy HF+ drivers. (see image of new SDRsharp)
[image]

These manual settings should be documented in the drivers, obviously, if they aren't there's nothing you can do.

Furthermore, given the rate of updated HF+ drivers, it may be best to wait a small while as the drivers mature.

Best wishes:

EB

csete, 2018-03-06:
Hi EB,

I am aware of these updates but I am very busy right now so no time to update. What I would like to know is how important these settings are and under which circumstances they help vs. the default AGC.

Alex

elibaum, 2018-03-07:
For the most part, I am not sure there is too extreme a difference with manual AGC vs what the HF uses by default.

I mentioned this feature more for completeness than functionality, the HF already has a very good default AGC. Obviously, although adding the manual settings is not a priority, it could be beneficial under some use-cases. It would be useful to have if/when you find time to add it.

I will continue to test, if I find anything shocking about manual AGC, I will let you know.

Much thanks,
Eli

gat3way, 2018-03-08:
As a workaround you may try this - https://github.com/gat3way/gr-osmosdr-gqrx/tree/airspyhf 

Exposed via gain settings (though far from ideal, but will work for most stuff)

elibaum, 2018-03-16:
Sorry for taking so long to respond. I was experiencing mac compilation blues. (dont do UHD kids)
gat3way's osmosdr seems to work as expected.

Much thanks, everyone,

EB

csete, 2020-01-26:
Airspy HF+ support is now available in upstream gr-osmosdr, though not yet merged into master. In any case, if gain controls are added to the gr-osmosdr backend, they will appear automatically in gqrx.

closed 2020-01-26 by csete