gqrx-sdr/gqrx #608: [Feature Request] Support for sound devices with sampling frequencies under 96 kHz as input devices

ea4eoz, 2018-03-08:
Hello.

I would love to use GQRX with Icom's 7100 USB sound card.  This transceiver has the posibility to output IF via its built in sound card but sampling frecuency is fixed at 48 kHz. I can do it using Soapy but it doesn't work well. It only works the first time you try, the second time GQRX hangs and the only way to see again the sound device is with a computer reset. Not very useful.

I thing support for input sound devices with sampling frequencies under 96 kHz directly in GQRX would be nice. This will allow to use any sound device as IQ input device.

Thanks for reading.

Miguel A. Vallejo, EA4EOZ

csete, 2018-03-08:
Hi Miguel,

Please understand that gqrx does not interface to any hardware directly and changing that would require rewriting a lot of code. It is not optimal, but it is what it is. I am now working on a new SDR project and would like to focus on that instead of keep spending time on gqrx. Due to the complex dependencies, gqrx is very frustrating to maintain and it is time to move on with a new design.

Alex

csete, 2018-03-08:
And just to comfort you, the new SDR will support softrock type receivers and transmitters, so it will readily support what you ask for here.

closed 2018-03-08 by csete

ea4eoz, 2018-03-09:
Glad to hear the good news!

Thank you.