gqrx-sdr/gqrx #609: GQRX Quit Unexpectedly

wholder, 2018-03-13:
I can run version 2.5.3, but not this new 2.9.1 version.  Crashes at startup with this info:

Wayne

Process:               gqrx [45540]
Path:                  /Applications/Gqrx 2.9.1-crashes.app/Contents/MacOS/gqrx
Identifier:            dk.gqrx.www
Version:               2.9 (0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           gqrx [45540]
User ID:               501

Date/Time:             2018-03-13 14:09:03.618 -0700
OS Version:            Mac OS X 10.12.6 (16G1212)
Report Version:        12
Anonymous UUID:        E15DD0F6-F382-54B7-5533-290069C395DE

Time Awake Since Boot: 1900000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff9603cd42 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff9612a457 pthread_kill + 90
2   libsystem_c.dylib             	0x00007fff95fa2420 abort + 129
3   libc++abi.dylib               	0x00007fff94afb94a abort_message + 266
4   libc++abi.dylib               	0x00007fff94b20c17 default_terminate_handler() + 243
5   libobjc.A.dylib               	0x00007fff9562f353 _objc_terminate() + 124
6   libc++abi.dylib               	0x00007fff94b1dd49 std::__terminate(void (*)()) + 8
7   libc++abi.dylib               	0x00007fff94b1d7be __cxa_throw + 121
8   libgnuradio-audio.3.7.10.1.dylib	0x00000001079006e4 gr::audio::osx_sink::setup() + 1060
9   libgnuradio-audio.3.7.10.1.dylib	0x00000001078ff7df gr::audio::osx_sink::osx_sink(int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) + 575
10  libgnuradio-audio.3.7.10.1.dylib	0x00000001078ff4a9 gr::audio::osx_sink_fcn(int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) + 57
11  libgnuradio-audio.3.7.10.1.dylib	0x00000001078dc89e gr::audio::sink::make(int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool) + 1550
12  dk.gqrx.www                   	0x0000000107062826 receiver::receiver(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, unsigned int) + 3638
13  dk.gqrx.www                   	0x000000010704f884 MainWindow::MainWindow(QString, bool, QWidget*) + 1172
14  dk.gqrx.www                   	0x000000010704a4db main + 5163
15  libdyld.dylib                 	0x00007fff95f0e235 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff9603d44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff96127621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9612707d start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff9603d44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff96127621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9612707d start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff9603d44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9612748e _pthread_wqthread + 1023
2   libsystem_pthread.dylib       	0x00007fff9612707d start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff9603d44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff96127621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9612707d start_wqthread + 13

[... 357 lines cut ...]
USB Device: USB Bus
USB Device: USB Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB Bus
USB Device: USB Bus
USB Device: USB Bus
USB Device: USB Bus
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB Audio CODEC
USB Device: Apple Cinema HD Display
USB Device: USB 2.0 Bus
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB2.0 Hub
USB Device: AVRISP mkII
USB Device: FT231X USB UART
USB Device: Blue Snowball
USB Device: USB2.0 Hub
USB Device: DYMO LabelManager PnP
USB Device: Logic S/16
FireWire Device: built-in_hub, Up to 800 Mb/sec
Thunderbolt Bus:

csete, 2018-03-13:
As I wrote both on the website, in the readme and the already existing OS X issues, I can not build for anything newer than 10.11.

closed 2018-03-13 by csete

csete, 2018-03-13:
Also, as far as I can tell you are using a Mac Pro and the problem is that the gnuradio audio backend wants a built-in audio device for default. I'm pretty sure there is another thread about this. Otherwise you can try running gqrx from a terminal to see what messages are printed and use the -r option to reset the configuration, which might let you choose another audio device.

Sorry I can't be of more help, but it's difficult for me to debug these things since I don't have representative hardware and OS.

wholder, 2018-03-13:
Thanks for taking the time to reply.  I was just passing along data in the
hopes that it might help.  I have an older version, 2.5.3, that seems to
work OK, so I'm using that for now.  I tried several of the suggested
workarounds, but could not seem to get them to work, but I'll continue
digging though the messages and see if I can try some other approaches.  I
tried running the app from the terminal, like this:

  open -a /Applications/Gqrx.app

and just got the same crash with no information.  I then tried adding on
-r, like this:

  open -a /Applications/Gqrx.app -r

but that confused the "open" program, which printed out a bunch of
suggestions.  I the tried using the 2.5.3 version to configure different
options followed by running the 2.9.1 version, but was unable to find a
setting that would prevent the crash.  If there is anything else I can do
to help try and diagnose this, please let me know.

Wayne

csete, 2018-03-13:
I can try to make a build using Portaudio as audio backend instead of gnuradio-audio, but since that option is only available on Linux right now, it will take some days to modify the code.

csete, 2018-03-14:
Can you try this dev build v2.10.10: http://files.oz9aec.net/gqrx/
I hope this audio backend works better... Other problems, like sometimes it is loading forever, remain.

Alex

wholder, 2018-03-15:
App launches and runs!

Wayne

csete, 2018-03-15:
Ok, thanks. I assume you also have audio.
I need to finish a few more things, then I will make an official release.

wholder, 2018-03-17:
Perhaps I'm not using it right, but I have run into some problems.  While
the app now starts up, I can't seem to get any audio out.  My SDR dongle is
a FightAware Prostick based on the R820T2 chip (although, it shows up
listed as "Realtek RTL2832UFA SN: 00001000" in the "Configure I/O Devices"
menu.)  I'm trying to use GQRX to debug an APRS transmitter I'm working
on.  I get the waterfall display and I can see the APRS transmits bursts
show up at 144,390 MHz and I can select this burst with the cursor to try
and listen to it.  But, no audio comes out.  Also, when I select the
AFSK1200 decoder option, the app crashes (log attached below.)  Actually,
the ability to use the AFSK1200 decoder is more important to me than audio
output, but I guessing the problems are related.

Wayne

Process:               gqrx [52853]
Path:                  /Applications/Gqrx 2.10.10-1.app/Contents/MacOS/gqrx
Identifier:            dk.gqrx.www
Version:               2.10.10 (0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           gqrx [52853]
User ID:               501

Date/Time:             2018-03-16 17:24:33.612 -0700
OS Version:            Mac OS X 10.12.6 (16G1212)
Report Version:        12
Anonymous UUID:        E15DD0F6-F382-54B7-5533-290069C395DE

Time Awake Since Boot: 2200000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
terminating with uncaught exception of type std::runtime_error:
pfb_arb_resampler_ccf(49): insufficient connected input ports (1 needed, 0
connected)

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib         0x00007fff9603cd42 __pthread_kill + 10
1   libsystem_pthread.dylib       0x00007fff9612a457 pthread_kill + 90
2   libsystem_c.dylib             0x00007fff95fa2420 abort + 129
3   libc++abi.dylib               0x00007fff94afb94a abort_message + 266
4   libc++abi.dylib               0x00007fff94b20c17
default_terminate_handler() + 243
5   libobjc.A.dylib               0x00007fff9562f353 _objc_terminate() + 124
6   libc++abi.dylib               0x00007fff94b1dd49 std::__terminate(void
(*)()) + 8
7   libc++abi.dylib               0x00007fff94b1d9d2 __cxa_rethrow + 99
8   libobjc.A.dylib               0x00007fff9562d6b7 objc_exception_rethrow
+ 40
9   com.apple.CoreFoundation       0x00007fff803445cb CFRunLoopRunSpecific
+ 555
10  com.apple.HIToolbox           0x00007fff7f8a3ebc
RunCurrentEventLoopInMode + 240
11  com.apple.HIToolbox           0x00007fff7f8a3bf9 ReceiveNextEventCommon
+ 184
12  com.apple.HIToolbox           0x00007fff7f8a3b26
_BlockUntilNextEventMatchingListInModeWithFilter + 71
13  com.apple.AppKit               0x00007fff7de3aa54 _DPSNextEvent + 1120
14  com.apple.AppKit               0x00007fff7e5b67ee
-[NSApplication(NSEvent)
_nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 2796
15  com.apple.AppKit               0x00007fff7de2f3db -[NSApplication run]
[... 1007 lines cut ...]
CoreAnimation                       24K        5
CoreUI image data                 1544K       12
CoreUI image file                  184K        5
Kernel Alloc Once                    8K        2
MALLOC                           454.0M       71
MALLOC guard page                   48K       10
MALLOC_LARGE (reserved)            188K        2         reserved VM
address space (unallocated)
Memory Tag 242                      12K        2
Memory Tag 249                     156K        3
OpenGL GLSL                        256K        4
STACK GUARD                       56.1M       28
Stack                             21.2M       30
VM_ALLOCATE                        236K       21
__DATA                            23.6M      280
__GLSLBUILTINS                    2588K        2
__IMAGE                            528K        2
__LINKEDIT                       123.5M       57
__TEXT                           167.0M      276
__UNICODE                          556K        2
mapped file                       75.9M       25
shared memory                     16.4M       13
===========                     =======  =======
TOTAL                            951.6M      853
TOTAL, minus reserved VM space   951.4M      853

csete, 2018-03-17:
I may take a look at the AFSK decode although I have publicly admitted that it is broken and should be removed. Even when it works, it's the shittiest decoder you can imagine. Please use Direwolf through the UDP interface or something else.

As for missing audio, I'm confused. I thought it worked? Are you sure it's not just some setting like squelch? A screenshot of the settings could help.

All these cheap dongles are based on the same chip set RTL2832U (R820T is the tuner). Gqrx can only see the chipset, not what sticker  the OEMs put on it.

wholder, 2018-03-17:
Sorry for not being clear.  When I first tried the app I didn't have the
SDR dangle handy (had to dig it out from where I'd stored it), so all I did
was fire up the app to see if it would start up and let me select an audio
output device.

Sad to hear the AFSK decoder is broken.  I'm not familiar with Direwolf,
but I'll see what I can learn about it and try to use that.  Does it run
on OS X?

Wayne

csete, 2018-03-17:
I would be very interested to know whether audio is working or not. I have the weekend to make a new release, after that I will be away for a long time.

wholder, 2018-03-17:
Ok.  I figured it out.  The "Mode" switch was set to "Demod Off".  Once I
selected something I began to get audio output.  I guess I need to learn a
bit more about how to use the app.

Wayne