gqrx-sdr/gqrx #611: Crash on startup

Rua, 2018-03-27:
Ever since the update 2.11.2, Gqrx no longer runs and crashes on startup. When I delete the default.conf file, this is the console output I get:

```
Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp plutosdr miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
Using audio backend: N/A
BookmarksFile is /home/rua/.config/gqrx/bookmarks.csv
getDeviceList : Available input devices:
   "Realtek RTL2838UHIDIR SN: 00000001"
   "PlutoSDR"
   "RFSPACE SDR-IQ Receiver"
   "RFSPACE SDR-IP Receiver"
   "RFSPACE NetSDR Receiver"
   "RFSPACE Cloud-IQ Receiver"
   "RTL-SDR Spectrum Server"
   "Red Pitaya Transceiver Server"
   "Complex Sampled (IQ) File"
Loading configuration from: "default.conf"
Configuration file: "/home/rua/.config/gqrx/default.conf"
No change in output device:
  old: 
  new: 
New FFT rate: 25 Hz
Requested bandwidth: 0 Hz
Actual bandwidth   : 0 Hz
setFreqCorr : 0 ppm
New LNB LO: 0 Hz
updateHWFrequencyRange failed fetching new hardware frequency range
New mode index: 2
Filter preset for mode 2 LO: -5000 HI: 5000
Generating taps for new filter   LO:-5000   HI:5000 TW:2000   Taps: 115
setFftRate to "25 fps"
New FFT rate: 25 Hz
New FFT rate: 25 Hz
setFftSize to "8192"
New FFT rate: 25 Hz
Launching I/O device editor
firstTimeConfig
CIoConfig : Available output devices:
    0 : "Built-in Audio Analog Stereo"
```
On subsequent runs, I get this instead:
```
Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp plutosdr miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
Using audio backend: N/A
BookmarksFile is /home/rua/.config/gqrx/bookmarks.csv
getDeviceList : Available input devices:
   "Realtek RTL2838UHIDIR SN: 00000001"
   "PlutoSDR"
   "RFSPACE SDR-IQ Receiver"
   "RFSPACE SDR-IP Receiver"
   "RFSPACE NetSDR Receiver"
   "RFSPACE Cloud-IQ Receiver"
   "RTL-SDR Spectrum Server"
   "Red Pitaya Transceiver Server"
   "Complex Sampled (IQ) File"
Loading configuration from: "default.conf"
Configuration file: "/home/rua/.config/gqrx/default.conf"
Crash guard triggered! 

```
I'm using an RTL-SDR dongle, but the same happens whether I have the dongle connected or not.

csete, 2018-03-27:
I'm sorry to hear that, but that's all I can tell you unless you provide me with details.

Rua, 2018-03-27:
What other details should I provide?

LongnoseRob, 2018-03-27:
@Rua 
How about the actual crash line?
Sofar we only know that you deleted the default.conf file..

Have you selected the right signal source and audio device in the configuration dialog?

Btw: you might get better support if you would check the mailing list/google group for GQRX first..
You can find more details http://gqrx.dk/user-support

Rua, 2018-03-27:
There is no configuration dialog, the program crashes as soon as I run it and nothing appears other than this console output.

csete, 2018-03-27:
> What other details should I provide?

We distribute gqrx in 11 different forms for different platforms. At the very least we need to know about your environment, like operating system, method of installation, etc. In this case it would also seem relevant to know which version of gqrx used to work for you and whether you can confirm that a binary for that version still works. For most platforms we now have standalone binary packages, so it is possible to run different versions without installation.

Rua, 2018-03-27:
Never mind. I restarted the system and suddenly it all works again. Really weird.

csete, 2018-03-27:
Ok, good :)

closed 2018-03-27 by csete