gqrx-sdr/gqrx #615: Lack of an option to change the audio output sample rate

mp107, 2018-04-03:
Welcome,
Although there is a drop-down list dedicated for changing an audio output sample rate the only available option is "48 kHz".
Is it possible to add another options, such as 192 kHz?
According to [this message](https://groups.google.com/forum/#!msg/gqrx/hWX4qebbIwM/De1S_nxrCQAJ]) it is possible to achieve that by modifying directly the source code and recompiling the application.
Is there any reason a way to change this setting has not been included in the application by default beside the one described in the linked message? Is it (from the code perspective) a lack of a standarized/general method to deduce the maximum supported sample rate for a particular soundcard?
Or is there any way to change this setting without source code modifications, for example by manual editing default.conf configuration file?

csete, 2018-04-04:
Hi @mp107 

Currently it is not possible to change the audio sample rate by editing the .conf files.

There is nothing technically impossible for supporting other audio rates. The reason why this isn't implemented is priority. When you have 100 items on the todo list but only have time to work on 5, you have to choose which 5 to work on and postpone the other 95 for later. It's that simple.

Gqrx has been available since 2011 and it is only now that people ask for other sample rates, so I think the prioritizing was not entirely wrong here ;-)

As I have explained in the email thread, increasing the audio rate to 192 kHz will not by itself give higher audio quality. The internal DSP will still run at around 48 kHz, as do most linux desktops unless reconfigured. So without further changes the only effect will be increased CPU consumption due to the extra resampling. The only are where other sample rates will have a direct benefit is if the audio system does not support 48 kHz, but so far I have not encountered any such system.

Anyway, I have no problems with having this as a feature request (I assume that's why you opened an issue here instead of continuing the discussion on the mailing list), but I will not be able to give this any high priority unless somebody can explain to me why this would be so important in the light of the above info.

OZ1SEJ, 2018-07-08:
I can understand why this is not at the top of your priority list :-) just want to add that when you record APT weather images from NOAA satellites and want to process the audio files into satellite images using APT3000, it expects them to be 11025 Hz sample rate, so you have to resample before uploading. That would be easier, if Gqrx could just record in 11025 Hz.

mp107, 2018-07-14:
@csete I understand.
For clarification, the reason I have asked about it is to allow to do the RDS decoding of received signal using an external application (in my case - RDS Spy as it has the Super PI detector feature which helps identifying received station, especially on E-sporadic openings).

darrenfreeman, 2020-12-10:
I have wished for this option for quite a while.

I am currently recording several days worth of audio as part of a scientific project. Even though I will have to down-convert it to 8 kHz, I am forced to record at 48 kHz. Also it's recording in stereo even though the demodulator is USB, a separate issue that doubles the pain.

I am nervous about the file size exceeding 4 GiB, as a WAV can't represent this size. Audacity only sees a subset of the data, when I go over. I will have to experiment with extracting the raw samples from the damaged files.

mercuriete, 2025-03-23:
Hi, I am using pipewire as audio stack.

There are any possibility that gqrx send the samples without resampling and pipewire takes responsability of that?

In my case I have a 96Khz audio interface and I saw this message:

```
Resampling audio 96000 -> 48000
```

In theory you could send to pulseaudio 96Khz and then pipewire would configure automatically my device to use 96Khz and avoid resampling.
I am using a focusrite scarlett 3i3 v3 capable of 96khz and 192khz as well as 48Khz and 44.1Khz.

Thanks for your work, amazing program.
:)

argilo, 2025-03-23:
> There are any possibility that gqrx send the samples without resampling and pipewire takes responsability of that?

There is no such option currently.

vladisslav2011, 2025-03-23:
> There are any possibility that gqrx send the samples without resampling and pipewire takes responsability of that?

You may take a look at this PR https://github.com/gqrx-sdr/gqrx/pull/1034