gqrx-sdr/gqrx #618: [Feature Request] Increase centre by one bandwidth.

elibaum, 2018-04-10:
I think it would be useful to be able to view one 'chunk' of bandwidth at a time. This would be useful both for large bandwidth devices so that you don't see the same signal twice, and small bandwidth devices so that you don't 'miss' any signals.

Let me know what you think?

Much thanks,

EB

csete, 2018-04-10:
Sorry, but I do not understand what you mean.

Sounds like your hardware is broken or incorrectly configured if you see the same signal twice.

elibaum, 2018-04-10:
Sorry, must have been unclear.

Imagine you are using an sdr stick with 2MHz of bandwidth. If I wanted to go from one 2MHz chunk to another, I need to click the area twice to view it. It would be easier to have a single button to automatically advance by 2MHz, rather than having to do it manually.

In this example, it may sound useless, but if you have a non-whole-number bandwidth, it can cause some of the issues above.

csete, 2018-04-10:
Ok, now I understand what you mean. Note that if we jump a whole number of bandwidth, we will be missing out stuff, because the usable alias free bandwidth is usually only 70-90% of the sampled bandwidth, depending on hardware device and settings. Just for info.

elibaum, 2018-04-10:
Maybe just jump the value of 70% each time, best to play it safe I guess.