gqrx-sdr/gqrx #625: Segfault on startup on macOS 10.12.6

samyk, 2018-04-28:
Hi!

I compiled from master (2838ca87c652beb1c9f6965b61388127cde3b725) though getting seg fault when running before a UI appears. macOS 10.12.6. Thoughts?

```
Process:               gqrx [90535]
Path:                  /Users/USER/*/gqrx
Identifier:            gqrx
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        bash [517]
Responsible:           gqrx [90535]
User ID:               501

Date/Time:             2018-04-28 14:20:57.867 -0700
OS Version:            Mac OS X 10.12.6 (16G1212)
Report Version:        12
Anonymous UUID:        ECD5F998-C1BE-EC0D-60A0-10CA7427E8CE

Sleep/Wake UUID:       D185C739-C875-4B32-8679-775096549A17

Time Awake Since Boot: 220000 seconds
Time Since Wake:       140000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0:
--> 
    __TEXT                 000000010a278000-000000010a419000 [ 1668K] r-x/rwx SM=COW  /Users/USER/*

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libc++.1.dylib                	0x00007fff9897b9b0 std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::operator=(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 14
1   gqrx                          	0x000000010a283191 boost::program_options::typed_value<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, char>::notify(boost::any const&) const + 97
2   libboost_program_options-mt.dylib	0x000000010a6ca873 boost::program_options::store(boost::program_options::basic_parsed_options<char> const&, boost::program_options::variables_map&, bool) + 1635
3   gqrx                          	0x000000010a27ecfe main + 1246
4   libdyld.dylib                 	0x00007fff99dad235 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff99edc44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff99fc6621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff99fc607d start_wqthread + 13

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff99fc6070 start_wqthread + 0
1   ???                           	0x000400080006000c 0 + 1125934266974220

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff99fc6070 start_wqthread + 0
1   ???                           	0x0000000400030002 0 + 17180065794

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff99fc6070 start_wqthread + 0

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff99edc44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff99fc6621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff99fc607d start_wqthread + 13

Thread 6:
0   libsystem_kernel.dylib        	0x00007fff99edc44e __workq_kernreturn + 10
[... 322 lines cut ...]
===========                     =======  ======= 
Accelerate framework               128K        2 
Activity Tracing                   256K        2 
CG backing stores                 1600K        4 
CG image                            36K        6 
CoreUI image file                  272K        4 
Dispatch continuations            16.0M        2 
Kernel Alloc Once                    8K        2 
MALLOC                            87.6M       20 
MALLOC guard page                   32K        6 
STACK GUARD                       56.0M        9 
Stack                             11.6M        9 
VM_ALLOCATE                         72K       10 
__DATA                            62.1M      296 
__GLSLBUILTINS                    2588K        2 
__IMAGE                            528K        2 
__LINKEDIT                       133.5M       74 
__TEXT                           215.3M      283 
__UNICODE                          556K        2 
mapped file                       54.0M       12 
shared memory                     16.3M       12 
===========                     =======  ======= 
TOTAL                            658.4M      739 

```

csete, 2018-04-28:
Well, it looks like boos::program_options used for command line parsing crashes. Do you have the same problem with the 2.11.4 app bundle? It could be that a different version of boost will work.

csete, 2018-04-29:
UHD is disabled in my binaries because there have been several problems and I am not able to test or fix, see e.g. https://github.com/csete/gqrx/issues/452
I don't know if it is the same problem you are seeing here, there are some hints at the end of the Ubuntu bug report: https://bugs.launchpad.net/ubuntu/+source/gqrx-sdr/+bug/1637774

argilo, 2020-11-15:
I removed boost:program_options in #850, so I doubt this is a problem anymore. Feel free to re-open if you're still seeing a crash.

closed 2020-11-15 by argilo