gqrx-sdr/gqrx #629: Add [space] keybinding

Arinerron, 2018-05-08:
Hello,

It'd be neat if there were a keybinding to toggle DSP processing, like the space bar. It's a simple change and in my opinion, it would make gqrx easier to use.

I'll make a PR if you'd like, but I am not sure I know exactly how to add it (I'm not a C++ dev).

Thanks,
    Aaron E. (aka: Arinerron)

LongnoseRob, 2018-05-08:
CTRL-D is the current shortcut

(Sent from my mobile device)

Arinerron, 2018-05-08:
Yup, I updated the issue on GitHub. :)

Sincerely,
    Aaron E. (aka: arinerron <https://www.arinerron.com/>)

csete, 2018-05-08:
Space bar is used by most graphical operating systems to activate the currently selected widget, which is an important part of making GUI applications usable for e.g. blind people. Sorry, but I prefer to stay with the existing short cut.

closed 2018-05-08 by csete