gqrx-sdr/gqrx #630: FATAL: destination port 1 out of range for source_impl(47)

KC7NOA, 2018-05-11:
Not sure if this is an old error or new   ...

Odroid-c2 Ubuntu 1604 64

kc7noaa@odroid64:~/Downloads/gqrx/build$ uname -r
3.14.79-117
********************************

FATAL: destination port 1 out of range for source_impl(47)

Trying to fill up 18446744073709551615 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

terminate called after throwing an instance of 'std::invalid_argument'
  what():  destination port 2 out of range for source_impl(47)
Aborted

KC7NOA, 2018-05-12:
Im not sure if thats true ...

Would SdrAngel use the same backend ?
And that issues is not there

csete, 2018-05-12:
> Im not sure if thats true ...

https://github.com/osmocom/gr-osmosdr/blob/master/lib/source_impl.cc#L432

KC7NOA, 2018-05-13:
Then how does the author of SdrAngel manage to avoid this problem

csete, 2018-05-13:
> Then how does the author of SdrAngel manage to avoid this problem

By not using GNU Radio and gr-osmosdr? Just  a wild guess ;)

KC7NOA, 2018-05-15:
Might be a good place to gleen some ideas from

csete, 2018-05-16:
> Might be a good place to gleen some ideas from

Gqrx uses GNU Radio to perform core DSP functionality and gr-osmosdr to get access to hardware devices. The application has been built up around these libraries and replacing them would mean practically rewriting the whole application.

SdrAngel, CubicSDR and Gqrx have taken different technical approaches to implement an SDR application. All three approaches have advantages and disadvantages; some will work better in some case, others will work better in other cases.