gqrx-sdr/gqrx #633: No waterfall visualization with USRP B200

celenkah, 2018-05-23:
Hello,
I have:
- gqrx 2.11.1-1 
- libuhd 3.10.2.0-3 
- gnuradio 3.7.11-5 
- boost 1.66.0-2 
- icu 61.1-1

As stated in the title, I cannot get a waterfall visualization while using gqrx with B200 ([screenshot](http://i.hizliresim.com/OD2093.png)); yet I can get it with RTL-SDR. It seems to be a similar issue with https://github.com/csete/gqrx/issues/145 ; but I cannot apply the solutions provided there since software versions mentioned are old. By the way, answers of that issue say that the problem has been fixed and the fix has been added to UHD 3.7.0; but somehow I still have this issue with UHD 3.10.

Thanks for any help; regards.

csete, 2018-05-23:
I can't see your screenshot, the server says forbidden. Please attach it here.

I don't understand your problem from your description, or at least I am confused by your reference to issue #145. Is your problem is only spectrum plot related or functional in general. For instance do you hear any audio? Do you see any messages printed in the terminal?

You also forgot to mention which operating system you are using and how you install the packages. Note that the latest version of gqrx is 2.11.5.

celenkah, 2018-05-23:
I'm sorry I didn't know how to attach here; here it is: [image]

I think problem is only related to the spectrum plot. I hear audio and there is no error in the terminal when it is run like `$ gqrx`. 

I'm using Arch Linux and installing packages using pacman; I only installed "icu60" from AUR which solved the "missing libicui18n.so.60" error (this error happened even I have installed icu v61 using pacman; it was something like v60 and v61 was both needed at the same time).

Latest version of gqrx in Arch repository is 2.11.1-1; but I will try to install 2.11.5 and report the results.

csete, 2018-05-23:
I think you just need to adjust the lower limit of the spectrum. The USRP is presumably a good receiver and it is not unlikely that your noise floor will be below -100 dBFS, also depending on FFT resolution.

You can adjust it by dragging or zooming on the dB-axis or using the sliders in the FFT tab.

celenkah, 2018-05-23:
Hi,
You are right, thank you! Sorry for not noticing this before.

closed 2018-05-23 by celenkah

csete, 2018-05-23:
No problem :)