gqrx-sdr/gqrx #640: Unable to build on Centos due to c++11 error

garengllc, 2018-06-08:
Will not build on Centos 7 as the tools seem to default to c++98 (though Ubuntu's tools are smart enough to know to build with c++11 enabled).  Centos can build the project fine if I add the line
`set(CMAKE_CXX_FLAGS "-std=c++11 -O3"` to the CMakeLists.txt file.

Could we add that as the standard to the file so that Centos users don't have to pybombs fetch, mod, and then rebuild?  It shouldn't effect the Ubuntu users (though I have no way of verifying that right now).

csete, 2018-06-08:
Ok, I will take a look when I can (very busy right now).

Note that -std=c++11 is gcc specific an may not work with other compilers, so it requires some more cmake code.

jimmystelzer, 2018-06-12:
@garengllc have you try to use `set (CMAKE_CXX_STANDARD 11)` in the CMakeLists.txt file? 
This is the compiler independent way to enable this feature under cmake.

garengllc, 2018-06-12:
Yes, and that works.  But as I mentioned in the first post, I was sort of hoping there could be some sort of fix in the CMakeLists.txt file that would cover this automagically.  Currently I pull it down, do a sed addition of the line and build it in my automated script I am running.  I was just hoping to remove that step.

I was thinking that it wouldn't hurt to just add that line, but csete pointed out that other compilers may not like it, so there is probably some wrapping that needs to go around that command (maybe a check to see if you are using gcc? I am not really sure).

csete, 2020-02-02:
Wondering whether this is still an issue?

garengllc, 2020-02-03:
Sorry, but I do not have access to this work anymore, so I am not really sure.  I think I was using sed in a setup script to make the mod and just went with that approach.
```
pybombs fetch
sed command
pybombs install
```

csete, 2020-02-03:
Okay, thanks for the update. Let's close it for now and see if the problem reappears in the future.

closed 2020-02-03 by csete