gqrx-sdr/gqrx #644: Waterfall doesn't move when the frequency at the top is changed

yurivict, 2018-06-24:
I change the frequency. The numbers on the scale change but the spectrum itself doesn't move.
So if there was some radiostation, after the move it is obviously misplaced.
Same with the 'Frequency' spin control. It changes the numbers, but doesn't move the spectrum.

2.11.5 on FreeBSD.

csete, 2018-06-24:
Your SDR device is probably broken. Please use the mailing list for user support, thanks.
http://gqrx.dk/user-support

closed 2018-06-24 by csete

yurivict, 2018-06-24:
I don't use a device. I use the new SpyServer support in https://github.com/racerxdl/gr-osmosdr

csete, 2018-06-24:
That's experimental at best. I barely had 5 minutes to test it. Anyway, the server is probably locked to not allow changing the hardware frequency.

yurivict, 2018-06-24:
Most likely that experimental project is still unstable, that's why, because all servers can't be locked.

racerxdl, 2018-06-25:
That's weird. Here it works just fine (only if I lock the server, then it doesnt change).
[image]

Can you provide a server address that you find the issue?

yurivict, 2018-06-25:
host=airspy.com,port=5557,spyserver=0

yurivict, 2018-06-25:
Why does it show ip=127.0.0.1 for you? Do you use a proxy?
-----
With host=airspy.com,port=5557,spyserver=0

yurivict, 2018-06-25:
[image]

racerxdl, 2018-06-25:
This server only streams from 6.9MHz~ to 7.720MHz~ so anything outside range will be empty.
[image]

PS: My picture showed localhost because the server is on my machine.

yurivict, 2018-06-25:
Its upper limit is 7.654MHz.