gqrx-sdr/gqrx #645: Changing megaheartz using a mouse wheel doesn't change frequency

yurivict, 2018-06-28:
[image]

gqrx-2.11.5 on FreeBSD 11.1

csete, 2018-06-28:
Please stop using this bug tracker for user support, we have a mailing list for that, see http://gqrx.dk/user-support.

Like I have already said in [my reply to your previous issue](https://github.com/csete/gqrx/issues/644#issuecomment-399798929), a spyserver can be locked in frequency so that you can only tune within its real-time bandwidth, in this case, 768 kHz.

Furthermore, spyserver support is currently limited to raw I/Q streaming which does not make sense over the Internet. Please use your local server before reporting issues.

closed 2018-06-28 by csete

yurivict, 2018-06-28:
> Please stop using this bug tracker for user support, we have a mailing list for that, see http://gqrx.dk/user-support.

> Like I have already said in my reply to your previous issue, a spyserver can be locked in frequency so that you can only tune within its real-time bandwidth, in this case, 768 kHz.

This looks and feels like a bug. I change frequency, and it doesn't change. Do you really think that this is a reasonable behavior that an average user would easily and intuitively understand?

If the frequency can't be changed, you should disallow the whole operation. Only allow operations that can be performed. Frequency obviously can be changed in other ways. I don't understand why scrolling GHz-es can't have the same effect.

amontefusco, 2018-06-28:
@yurivict

gqrx interacts with hardware using gr-osmosdr. it can do everything the several classes (one per radio hardware) allow to do, nothing more, nothing less.

Eventually, the gr-osmosdr fork for airpsy server should be blamed.

amontefusco, 2018-06-28:
Shouldn't it signal that, server-imposed, limitation using the frequency get/set methods return values?
( supposing the client-server protocol is able to vehiculate that information)

csete, 2018-06-28:
@amontefusco 

gr-osmosdr seems to return the correct frequency range. That's why the frequency controller in gqrx appears "frozen" when the user tries to tune outside that range. At least that's what happens with the [binary I made available yesterday](https://sourceforge.net/projects/gqrx/files/testing/), but I have no idea what code @yurivict uses.

Okay guys, this has been fun, but I am a busy person and don't have time for this. Please, no more bug reports on gqrx unless it is confirmed with something else than spyserver.