gqrx-sdr/gqrx #646: UDP streaming is not working

moskk, 2018-07-05:
hello, i'm running Gqrx 2.11.1 on archlinux 4.17.3-1-ARCH. 
1. run _sudo gqrx_, connect to SDR
2. set low squelch level to have some sound.
3. run _sudo netstat -naup_ in terminal:
```
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
udp        0      0 0.0.0.0:27036           0.0.0.0:*                           4235/steam          
udp        0      0 0.0.0.0:68              0.0.0.0:*                           1361/dhcpcd         
udp6       0      0 ::1:56946               ::1:56946               ESTABLISHED 511/postgres        
udp6       0      0 :::41235                :::*                                1294/Preload.js --c 
udp6       0      0 fe80::eca1:28df:76c:546 :::*                                1361/dhcpcd   
```
4. click "Stream raw audio over UDP" button to start streaming, check netstat:
```
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
udp        0      0 0.0.0.0:37519           0.0.0.0:*                           15643/gqrx          
udp        0      0 0.0.0.0:27036           0.0.0.0:*                           4235/steam          
udp        0      0 0.0.0.0:68              0.0.0.0:*                           1361/dhcpcd         
udp6       0      0 ::1:56946               ::1:56946               ESTABLISHED 511/postgres        
udp6       0      0 :::41235                :::*                                1294/Preload.js --c 
udp6       0      0 fe80::eca1:28df:76c:546 :::*                                1361/dhcpcd   
```
5. click "Stream raw audio over UDP" button again to stop streaming, check netstat:
```
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
udp        0      0 0.0.0.0:27036           0.0.0.0:*                           4235/steam          
udp        0      0 0.0.0.0:68              0.0.0.0:*                           1361/dhcpcd         
udp6       0      0 ::1:56946               ::1:56946               ESTABLISHED 511/postgres        
udp6       0      0 :::41235                :::*                                1294/Preload.js --c 
udp6       0      0 fe80::eca1:28df:76c:546 :::*        
```
6. repeat steps 4 and 5:
```
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
udp        0      0 0.0.0.0:27036           0.0.0.0:*                           4235/steam          
udp        0      0 0.0.0.0:68              0.0.0.0:*                           1361/dhcpcd         
udp        0      0 0.0.0.0:49955           0.0.0.0:*                           15643/gqrx          
udp6       0      0 ::1:56946               ::1:56946               ESTABLISHED 511/postgres        
udp6       0      0 :::41235                :::*                                1294/Preload.js --c 
udp6       0      0 fe80::eca1:28df:76c:546 :::*        
```
```
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
udp        0      0 192.168.88.252:38093    217.118.84.4:443        ESTABLISHED 1441/openconnect    
udp        0      0 0.0.0.0:27036           0.0.0.0:*                           4235/steam          
udp        0      0 0.0.0.0:68              0.0.0.0:*                           1361/dhcpcd         
udp6       0      0 ::1:56946               ::1:56946               ESTABLISHED 511/postgres        
udp6       0      0 :::41235                :::*                                1294/Preload.js --c 
udp6       0      0 fe80::eca1:28df:76c:546 :::*    
```
looks like gqrx binds to some ports when streaming is enabled, but _nc -l -u_ on them don't produce any output. and also port numbers are changing from time to time while it is unchanged in audio settings dialog:
[image]
i tried to set host address to localhost and ::ffff:127.0.0.1, it does no visible changes.
i saved the output of _sudo strace gqrx 2>gqrx.strace_ command while i did the steps described above, you can find it [here](https://yadi.sk/d/u5lpu5s03YoRmB). without strace (launched as _sudo gqrx_) the program behaves similarly.

moskk, 2018-07-06:
alexf91 thanks for response. you right, gqrx sent data to ::1.

closed 2018-07-06 by moskk

davideserra, 2021-11-12:
Hi guys, 
This situation is driving me crazy. I have a raspberry PI 4 with GPRX installed on and it work properly, but all my effort to stream audio on UDP was falied. I have to consume that audio from windows (may be vlc? any other audio software for me was good) but all setting I did was failed. Really I didn't understood where I'm wrong. Jiust by now, after setting IPV6 string on Raspberry GQRX pointing on my windows OS, wile I'm writing this comment, every now and then I hear some "noise" (I am listening to a radiotelescope so hearing noise is what I wont to hear) but seem to me that the system is working as like as hiccup. Any suggestion will be appreciate. Thanks so much.