gqrx-sdr/gqrx #649: gqrx 2.11.5 crashes on Ubuntu 18.04

averri, 2018-08-05:
I did a clean install with:

```
sudo apt-get purge --auto-remove gqrx
sudo apt-get purge --auto-remove gqrx-sdr
sudo apt-get purge --auto-remove libgnuradio*

sudo add-apt-repository -y ppa:bladerf/bladerf
sudo add-apt-repository -y ppa:myriadrf/drivers
sudo add-apt-repository -y ppa:myriadrf/gnuradio
sudo add-apt-repository -y ppa:gqrx/gqrx-sdr
sudo apt-get update

sudo apt-get install gqrx-sdr
```

And the gqrx is crashing. When it opens, after connecting the HackRF and pressing the start button it does not show the RF spectrum.

Please inform where the logs are located.

alexf91, 2018-08-09:
Gqrx logs everything to stdout. Start it in a terminal and copy/paste the output here.

csete, 2018-08-09:
I just tried a fresh install on Ubuntu 18.04 and it works fine with the hackrf. Make sure you use the gr-osmosdr backend, not the soapysdr. Also make sure you run the application after you connected the hackrf.

Technical support is provided through the mailing list, see http://gqrx.dk/user-support - let's keep this place for tracking bugs and feature requests, thanks.

closed 2018-08-09 by csete