gqrx-sdr/gqrx #654: [Feature] 'am unselective'

deckbsd, 2018-09-05:
Hello,

I saw in video [on youtube](https://www.youtube.com/watch?v=psuEzxFJnZY) ( time : 13:15) of a gnu radio receiver that automaticaly catch the biggest amp signal and select it. It could be really cool for example the airband. Of course this will only be available with AM modulation, but i think it could be great. I don't know if i explained it clearly, but here are some few links about it : 

https://kpreid.livejournal.com/
https://github.com/kpreid/shinysdr

Unfortunatly i'm not comfortable enougth with gnu radio and math to do it myself :/

Don't hesitate to ask for more informations, i wrote this issue very quickly.

amontefusco, 2018-09-06:
What has your question had to do with gqrx?

deckbsd, 2018-09-07:
Because gqrx is powered by gnu radio and the guy did it with gnu radio, so it should be compatible ... so it could be cool to have this capability inside gqrx instead of running again an other software aside for one capability. But if it s not possible so ok.

amontefusco, 2018-09-07:
It is available in theory, he has just  to add a little bit of c++ code in order to integrate any GR module he needs.