gqrx-sdr/gqrx #658: Is gqrx unmaintained?

Keziolio, 2018-10-03:
If so, @csete can you name another maintainer and/or update the project status on the readme?
There are a bunch of great PR to merge, apart from my very minor one (that has gone ignored too).
I read that there are no future feature release planned, and that you have another sdr project in mind, I see "softrig", I built it, but it's very raw still

gqrx is the only decent sdr receiver that we have outside windows, can we please not let it die in the months / years that will take to make a viable replacement?

Thanks

csete, 2018-10-03:
Why would it die? It compiles and works well, if not we will look into it. Please accept the fact, that people who do these projects in their spare time may choose to have a life outside these projects from time to time.

closed 2018-10-03 by csete

Kezii, 2018-10-03:
You have probably misunderstood me

It's absolutely right that you have better things to do, that's the point, I'm not saying that you should do anything, i'm only proposing that maybe this project can be set up in a manner that _someone else_ do all the heavy lifting for you, i'm sure there are a lot of people interested in contributing new features and maintain the software

https://medium.com/@shazow/how-to-hand-over-an-open-source-project-to-a-new-maintainer-db433aaf57e8

Because, as a gqrx user I wanted for a long time some of the things that are already implemented in the PRs, obviously I can fork it and merge the stuff I want but that's not helping anyone else, and since this is the only usable sdr software for a lot of users I thought about raising the issue up

csete, 2018-10-04:
I understand you very well, I just get tired of having these discussions all the time. I know you mean well, but you should do a bit more background check about people. If you did that with me, you would probably find it unnecessary to send me a link about how to hand over an open-source project as I have already done that several times over the last 20 years ;)

The problem is indeed that many people are interested in contributing new features, but that is only 5% of the work. There is much more to software engineering than adding new features, but not many people are interested in doing the remaining 95% of the work. I have been trying to find help with that for years, but no luck. That is why I am not interested in adding new features for now.

So let's just end this thread by letting you know that we are actively looking for new maintainers.