gqrx-sdr/gqrx #659: Running gqrx with xvfb fails

Sevyls, 2018-10-05:
I tried to run gqrx with xvfb to simulate a DISPLAY on the CLI for usage with remote control and udp sink.

When I start it by running
`xvfb-run gqrx`

I get the following error and abort:

```
$ xvfb-run gqrx
linux; GNU C++ version 6.2.0 20161010; Boost_106100; UHD_003.009.005-0-unknown

libEGL warning: DRI2: failed to create any config
libEGL warning: DRI2: failed to create any config
Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
FM demod gain: 1.52789
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
QPainter::begin: Paint device returned engine == 0, type: 2
QPainter::initFrom: Painter not active, aborted
QPainter::setFont: Painter not active
QPainter::setBrush: Painter not active
QPainter::font: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setOpacity: Painter not active
QPainter::setOpacity: Painter not active
QPainter::setPen: Painter not active
QPainter::end: Painter not active, aborted
QPainter::begin: Paint device returned engine == 0, type: 2
QPainter::initFrom: Painter not active, aborted
QPainter::setFont: Painter not active
QPainter::setBrush: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::end: Painter not active, aborted
BookmarksFile is /home/pi/.config/gqrx/bookmarks.csv
*** Error in `gqrx': corrupted double-linked list: 0x01469498 ***
Aborted

```

I then tried another config and another error resulted in an abort:

```
$ xvfb-run gqrx -c ~/pocsag.conf 
linux; GNU C++ version 6.2.0 20161010; Boost_106100; UHD_003.009.005-0-unknown

libEGL warning: DRI2: failed to create any config
libEGL warning: DRI2: failed to create any config
Controlport disabled
User specified config file: "/home/pi/pocsag.conf"
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
FM demod gain: 1.52789
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
QPainter::begin: Paint device returned engine == 0, type: 2
QPainter::initFrom: Painter not active, aborted
QPainter::setFont: Painter not active
QPainter::setBrush: Painter not active
QPainter::font: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setOpacity: Painter not active
QPainter::setOpacity: Painter not active
QPainter::setPen: Painter not active
QPainter::end: Painter not active, aborted
QPainter::begin: Paint device returned engine == 0, type: 2
QPainter::initFrom: Painter not active, aborted
QPainter::setFont: Painter not active
QPainter::setBrush: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::setPen: Painter not active
QPainter::end: Painter not active, aborted
BookmarksFile is /home/pi/.config/gqrx/bookmarks.csv
*** Error in `gqrx': malloc(): smallbin double linked list corrupted: 0x00d44c70 ***
Aborted
```

Any ideas?

alexf91, 2018-10-05:
This linked list error has been reported a few times, and I don't think we know what causes it. It only happens on the Raspberry Pi or other ARM boards.
It usually works if you install Gqrx from here:
http://gqrx.dk/download/gqrx-sdr-for-the-raspberry-pi

How did you install Gqrx?

alexf91, 2018-10-05:
The Raspbian package is probably outdated. They used to package 2.6, which is almost two years old. The version from the website should work.

Sevyls, 2018-10-05:
Thanks.
I actually can't access gqrx.dk right now,

checked from multiple connections including

https://downforeveryoneorjustme.com/gqrx.dk
with "It's not just you! gqrx.dk looks down from here."

alexf91, 2018-10-05:
This is the same archive:
https://github.com/csete/gqrx/releases/download/v2.11.5/gqrx-sdr-2.11.5-linux-rpi3.tar.xz

Sevyls, 2018-10-05:
Thank you, this version works.

closed 2018-10-05 by Sevyls