gqrx-sdr/gqrx #660: upgrading libgnuradio-osmosdr0.1.4 on xenial

highflux7, 2018-10-18:
During usual system updates on ubuntu xenial (16.04), get:

`sudo apt list --upgradeable -a`

Listing...
Done
libgnuradio-osmosdr0.1.4/xenial 0.1.4.127-myriadrf2~xenial amd64 [upgradable from: 0.1.4.98-myriadrf17~xenial]
libgnuradio-osmosdr0.1.4/now 0.1.4.98-myriadrf17~xenial amd64 [installed,upgradable to: 0.1.4.127-myriadrf2~xenial]
libgnuradio-osmosdr0.1.4/xenial 0.1.4-8 amd64

Subsequently,
`sudo apt-get upgrade libgnuradio-osmosdr0.1.4`
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:Listing...
 libgnuradio-osmosdr0.1.4 : Depends: libbladerf2 but it is not installable
E: Broken packages

Suggestionos?

(apologies for the poor format-fu)

alexf91, 2018-10-18:
If you don't need bladeRF support, then use the binary package for Gqrx from Github. It is the latest release of Gqrx and includes a recent version of gr-osmosdr and libraries for many devices.

If you want to use it with the bladeRF, then we have to find a different solution. Recent bladeRF releases don't work with the 4 year old gr-osmosdr 0.1.4 version, and the binary release of Gqrx does not include the bladeRF libraries.

csete, 2018-10-18:
Not sure what you are trying to do, "usual system updates" for me is just "sudo apt upgrade" without any specific package at the end.

There is a convoluted list of dependencies. Therefore, it is important to know what and how you have installed in the past. If you used PPA, please make sure you still have all PPAs enabled since the bladerf dependency comes from the bladerf PPA.

As a last resort, you can try to disable PPA, remove gqrx and its dependencies and reinstall following http://gqrx.dk/download/install-ubuntu

closed 2018-10-18 by csete