gqrx-sdr/gqrx #661: gqrx gr-osmosdr-gqrx airspyhf branch does not compile

Sevyls, 2018-10-19:
Trying to build gqrx from source with airspyhf support.

Came to the Repo with the patched gr-osmosdr-gqrx branch https://github.com/csete/gr-osmosdr-gqrx/tree/airspyhf 

Will not compile due to the following errors.

rtl_tcp_source_c.cc

Any ideas why I get a compile error?

```
$ make
Scanning dependencies of target gnuradio-osmosdr
[  7%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o
[ 14%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/sink_impl.cc.o
[ 21%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/ranges.cc.o
[ 28%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/device.cc.o
[ 35%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/time_spec.cc.o
[ 42%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/file/file_source_c.cc.o
[ 50%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/file/file_sink_c.cc.o
[ 57%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/rtl_tcp/rtl_tcp_source_c.cc.o
/Users/user/Projekte/gr-osmosdr-gqrx/lib/rtl_tcp/rtl_tcp_source_c.cc:301:9: error: 
      expected expression
  cmd = { 0x09, htonl(direct_samp) };
        ^
/Users/user/Projekte/gr-osmosdr-gqrx/lib/rtl_tcp/rtl_tcp_source_c.cc:307:9: error: 
      expected expression
  cmd = { 0x0a, htonl(offset_tune) };
        ^
/Users/user/Projekte/gr-osmosdr-gqrx/lib/rtl_tcp/rtl_tcp_source_c.cc:311:9: error: 
      expected expression
  cmd = { 0x0e, htonl(bias_tee) };
        ^
/Users/user/Projekte/gr-osmosdr-gqrx/lib/rtl_tcp/rtl_tcp_source_c.cc:570:9: error: 
      expected expression
  cmd = { 0x08, htonl(automatic) };
        ^
4 errors generated.
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/rtl_tcp/rtl_tcp_source_c.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
make: *** [all] Error 2

```

```

$ gcc -v
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
Apple LLVM version 10.0.0 (clang-1000.10.44.2)
Target: x86_64-apple-darwin18.0.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

csete, 2018-10-19:
[Like I said](https://github.com/csete/gr-osmosdr-gqrx/issues/11#issuecomment-431335066), please try to build the upstream [gr-osmosdr](https://github.com/osmocom/gr-osmosdr) first to see if that works. Neither gr-osmosdr-gqrx nor gqrx has anything to do with your issues.

closed 2018-10-19 by csete

Sevyls, 2018-10-19:
> Neither gr-osmosdr-gqrx nor gqrx has anything to do with your issues.

I disagree - as gqrx comes with builtin airspyhf support and I am unable to build it from source with the same device support as delivered with the gqrx binaries.

I get the same errors with the master from gr-osmosdr.

alexf91, 2018-10-19:
This is related to your compiler or C/C++ version.

csete, 2018-10-19:
> as gqrx comes with builtin airspyhf support and I am unable to build it from source with the same device support as delivered with the gqrx binaries.

Gqrx does not come builtin with any hardware support - it is just a GUI. It uses an external library, gr-osmosdr, to provide hardware support. 

> I get the same errors with the master from gr-osmosdr.

Good. So the problem is with gr-osmosdr.