gqrx-sdr/gqrx #667: AppImage missing qt plugin xcb

teintuc, 2018-12-23:
Hi,

I have been trying to play with gqrx ussing the AppImage 2.11.5.
Unfortunatly, I get this error message:

<pre>This application failed to start because it could not find or load the Qt platform plugin "xcb".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, xcb.

Reinstalling the application may fix this problem.
zsh: abort (core dumped)  ./Gqrx-2.11.5-x86_64.AppImage</pre>

I don't really know how to fix it. I have been looking for the missing package without success. I'll keep digging but if somebody has an idea!

Fyi: I'm on archlinux

Thanks

csete, 2018-12-23:
While you are welcome to try an find out whether this is a problem with AppImage, I want to point out that the latest version of gqrx is already available in archlinux.

AsciiWolf, 2019-10-05:
I have exactly the same issue on Fedora 30.

AsciiWolf, 2019-10-21:
Would it be possible to reopen this issue?

AsciiWolf, 2019-11-08:
Here are some logs when running the Gqrx AppImage with `QT_DEBUG_PLUGINS=1` on Fedora 30 (after installing _libnsl_):
```
QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms" ...
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqeglfs.so"
Found metadata in lib /tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqeglfs.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "eglfs"
        ]
    },
    "className": "QEglFSIntegrationPlugin",
    "debug": false,
    "version": 328961
}

Got keys from plugin meta data ("eglfs")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqlinuxfb.so"
Found metadata in lib /tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqlinuxfb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "linuxfb"
        ]
    },
    "className": "QLinuxFbIntegrationPlugin",
    "debug": false,
    "version": 328961
}

Got keys from plugin meta data ("linuxfb")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqminimal.so"
Found metadata in lib /tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqminimal.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimal"
        ]
    },
    "className": "QMinimalIntegrationPlugin",
    "debug": false,
    "version": 328961
}

Got keys from plugin meta data ("minimal")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqminimalegl.so"
Found metadata in lib /tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqminimalegl.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimalegl"
        ]
    },
    "className": "QMinimalEglIntegrationPlugin",
    "debug": false,
    "version": 328961
}

Got keys from plugin meta data ("minimalegl")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqoffscreen.so"
Found metadata in lib /tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqoffscreen.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "offscreen"
        ]
    },
    "className": "QOffscreenIntegrationPlugin",
    "debug": false,
    "version": 328961
}

Got keys from plugin meta data ("offscreen")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so"
Found metadata in lib /tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "xcb"
        ]
    },
    "className": "QXcbIntegrationPlugin",
    "debug": false,
    "version": 328961
}

Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_Gqrx-2mrMVB5/usr/bin/platforms" ...
loaded library "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so"
QLibraryPrivate::loadPlugin failed on "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so" : "Cannot load library /tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so: (/lib64/libfontconfig.so.1: undefined symbol: FT_Done_MM_Var)"
This application failed to start because it could not find or load the Qt platform plugin "xcb".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)
```

This seems to be the problem:
```
QLibraryPrivate::loadPlugin failed on "/tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so" : "Cannot load library /tmp/.mount_Gqrx-2mrMVB5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so: (/lib64/libfontconfig.so.1: undefined symbol: FT_Done_MM_Var)"
```

csete, 2019-11-08:
Thanks for the update @AsciiWolf :)

For information, the AppImage is built based on the Ubuntu PPA and here are my notes about the method that I used to create the AppImage:

Source: https://github.com/AppImage/AppImages

    git clone https://github.com/AppImage/AppImages.git appimages.git
    cd appimages.git
    bash -ex ./pkg2appimage recipes/Gqrx.yml
    cd ..
    wget https://github.com/AppImage/AppImageKit/releases/download/continuous/appimagetool-x86_64.AppImage
    chmod +x appimagetool-x86_64.AppImage
    cp -r appimages.git/Gqrx/Gqrx.AppDir .

Need to remove these to avoid symbol lookup error, otherwise modify .yml to use a Xenial (16.04) or Binoic (18.04):

    rm Gqrx.AppDir/lib/x86_64-linux-gnu/libglib-2.0.so.0 
    rm Gqrx.AppDir/lib/x86_64-linux-gnu/libglib-2.0.so.0.4000.0 
    rm Gqrx.AppDir/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 
    rm Gqrx.AppDir/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4000.0 

Finally:

    ./appimagetool-x86_64.AppImage Gqrx.AppDir Gqrx-2.11.2-x86_64.AppImage

csete, 2020-04-18:
There is now a new working AppImage for the 2.12.1 release.

closed 2020-04-18 by csete