gqrx-sdr/gqrx #668: Gqrx not running on Archlinux

teintuc, 2018-12-29:
Hi,

I followed the advice from #667 and I installed the gqrx 2.11.5 from the repository.
Unfortunatly, the application is really really slow just to show me the configuration window.
In a cli I just get these messages:
``` 
$gqrx
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Resampling audio 96000 -> 48000
BookmarksFile is /home/miky/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 8.2.1 20180831; Boost_106800; UHD_3.13.0.1-0-unknown
```

When I finally get the configuration window, the main one won't show up.
I'm new into the SDR world so any help is welcome.

Thanks

alexf91, 2018-12-29:
# Everything is broken...

The whole release process of Gqrx (and all other programs relying on `gnuradio-osmosdr`) is kind of broken for most (all?) distributions, because they ship the latest "stable" release of `gnuradio-osmosdr` (0.1.4), which is almost 5 years old. This is going to get worse with GNU Radio 3.8, but that's another story.

## ...but it's not as bad as you think

You can start with the [gqrx-sdr-2.11-5](https://github.com/csete/gqrx/releases/download/v2.11.5/gqrx-sdr-2.11.5-linux-x64.tar.xz) image from Github. It provides all the necessary libraries to run Gqrx, including a recent version of `gnuradio-osmosdr`, you just have to copy the udev rules to `/etc/udev/rules.d/`. You can run it from the extracted archive with `./gqrx`.

This should work on all distributions.

## Installing from the AUR

You can also build Gqrx from the AUR. Install `gr-osmosdr-git` from the AUR and then install `gqrx-git` with your favorite AUR-helper (I use [yay](https://aur.archlinux.org/packages/yay/)). This essentially results in the same build as the archive from Github or http://gqrx.dk.

On Arch Linux, I recommend the installation from the AUR.

teintuc, 2018-12-30:
Hi,

Thanks for your help. Do you have any link I could read to learn more about all this?

I tried both solutions:
- The image one is perfect and run out of the box, as you said. Works with an rtl-sdr but crash with my limesdr mini (probably an other problem)
- I also tried your recommendation: installed gr-osmosdr-git and gqrx-git from aur but it is still really slow to start. It worked for the rtl-sdr but also crash for the limesdr-mini

```
> gqrx
gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri rfspace airspy soapy redpitaya 
Resampling audio 96000 -> 48000
BookmarksFile is /home/miky/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 8.2.1 20180831; Boost_106800; UHD_3.13.0.1-0-unknown
gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri rfspace airspy soapy redpitaya 
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D3BD157C17A71'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M calibration values caching Disable
[INFO] RX LPF configured
terminate called after throwing an instance of 'std::runtime_error'
  what():  In hierarchical block source_impl, output 1 is not connected internally
zsh: abort (core dumped)  gqrx
```

tomswartz07, 2018-12-31:
@teintuc I also ran into this problem and @alexf91's second solution is the one to use.

Specific to your LimeSDR Mini issue (which I also came across), you need to install the `aur/soapylms7-git` package prior to building `gqrx-git`. 

It's worth mentioning that I get best performance using Soapy Drivers for LimeSDR, as described in the docs: http://gqrx.dk/supported-hardware#limesdr

Typically, my update process for GQRX and the toolchain goes something like this:
1. Install/update SoapySDR plugins
1. Install/upgrade `gr-osmosdr-git`
1. Install/upgrade `gqrx-git`

Then my device string is `driver=lime,soapy=0`

In the `.config/gqrx/default.conf` file, here's my input block that works:
```ini
[input]
antenna=LNAW
dc_cancel=true
decimation=2
device="driver=lime,soapy=0"
frequency=147015000
gains=@Variant(\0\0\0\b\0\0\0\x3\0\0\0\x6\0T\0I\0\x41\0\0\0\x2\0\0\0x\0\0\0\x6\0P\0G\0\x41\0\0\0\x2\0\0\0\x61\0\0\0\x6\0L\0N\0\x41\0\0\0\x2\0\0\0\xfe)
sample_rate=9500000
```

teintuc, 2019-01-01:
@tomswartz07 Thank you so much for your help. It works fine now. 
It still take some time to start but I'm fine with it.

Thanks again

closed 2019-01-01 by teintuc