gqrx-sdr/gqrx #672: multiple waterfalls per device

drom, 2019-01-12:
When working in HAM radio contests it is possible to capture all band by a single dongle but it is convenient to see multiple waterfalls, one per sub-band.

For example 144MHz band (1.5MHz) as 3 waterfalls:
  * 144.040-144.150
  * 144.280-144.350
  * 145.450-145.550

Is this possible in GQRX?

eu2aa, 2019-01-12:
it's impossible

drom, 2019-01-12:
@csete how difficult would be to implement such feature?

csete, 2019-01-12:
@drom I no longer work on gqrx as it has become unfeasible to maintain it. It's just broken at the architectural level and the only way to fix it IMO is to rewrite most of it from scratch, which is ongoing.

closed 2019-01-12 by csete

drom, 2019-01-12:
Thank you for your reply @csete

csete, 2019-01-12:
I can tell you that doing a quick hack with two or more spectrum displays could  be done relatively easily, as long as you don't want multiple receivers/demodulators running in parallel.

alexf91, 2019-01-12:
I have just implemented this hack:
https://github.com/alexf91/gqrx/tree/multiple-waterfalls
[image]