gqrx-sdr/gqrx #673: [Feature Request]

handiko, 2019-01-15:
Hi, I am Handiko from Indonesia.

I do really like Gqrx, especially with the presence of APRS / AFSK1200 decoder. The decoding performance is good enough for my circumstances. It could decode my portable (!) APRS tracker using 100mW (!) of power from about 5 km away with helical antenna, and RTL-SDR + 7 m high ground plane antenna at the receiver. 

Is it possible to add I-Gate functionality with it? Sure there is Direwolf, but what I really like from Gqrx is that it has the FFT display, unlike the Direwolf which doesn't. 

Thanks before.

handiko

kukabu, 2019-01-15:
Hello
It's already possible.
http://gqrx.dk/doc/streaming-audio-over-udp

handiko, 2019-01-15:
> Hello
> It's already possible.
> http://gqrx.dk/doc/streaming-audio-over-udp

Oh my god. I'm sorry, I didn't know that. 
Thanks. I already try it and it works very well.

Just ignore this feature request, then.
Sorry.

csete, 2019-01-15:
No problem.

closed 2019-01-15 by csete