gqrx-sdr/gqrx #675: gqrx shouldn't exit with SIGABRT on user errors

l29ah, 2019-01-31:
Say,
```
[blah]
Using device #0 Generic RTL2832U OEM
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to open rtlsdr device.
zsh: abort (core dumped)  gqrx
```
shouldn't happen as it is not a programming error, yet it produces an useless core dump.

csete, 2019-01-31:
This is a weakness of the architecture, but please do send a pull request if you think it can be fixed in a clean way.

argilo, 2020-02-05:
This issue can be closed, since it was addressed in 34333e78b2e565745b6d62bf0efcc57cc3b1988f.

closed 2020-02-05 by l29ah