gqrx-sdr/gqrx #676: Hardware AGC has to be reset when starting GQRX

qualitymanifest, 2019-01-31:
- Operating system: Linux Mint 19
- GQRX version: 2.9
- Installed from: Synaptic Package Manager
- Device: RTL-SDR v3 dongle

Every time I start GQRX I have to un-check and re-check "Hardware AGC". I get no reception until I do and the waterfall display looks completely different afterwards, as shown in the screenshot provided. It works totally fine after I reset it.

I found this related issue https://github.com/csete/gqrx/issues/370 that says it was fixed, but my GQRX version was released long after that issue was said to have been resolved.

Thanks in advance, I really appreciate this software!

[image]

csete, 2019-01-31:
Please try 2.11! We have an Ubuntu PPA, a generic tar.gz binary, and an AppImage on the website, the latter two will not even require uninstalling 2.9 see http://gqrx.dk/download

csete, 2019-01-31:
Works for me with gqrx 2.11 and Qt 5.6

closed 2019-01-31 by qualitymanifest