gqrx-sdr/gqrx #683: gqrx: tpp.c:84: __pthread_tpp_change_priority

PA8E, 2019-04-08:
testing the new build of GQRX. After connecting to spyserver and giving about 2 seconds of audio, GQRX crashes with: 

/usr/local/bin/gqrx -r
gr-osmosdr v0.1.4-144-g47a39bad (0.1.5git) gnuradio 3.7.11
built-in source types: file fcd rtl_tcp rfspace airspy redpitaya spyserver 
Resampling audio 96000 -> 48000
gr-osmosdr v0.1.4-144-g47a39bad (0.1.5git) gnuradio 3.7.11
built-in source types: file fcd rtl_tcp rfspace airspy redpitaya spyserver 
SpyServer: Trying to connect
SpyServer: Connected
SpyServer: Got sync Info
SpyServer: Maximum Sample Rate: 2400000
SpyServer: Ready
SpyServer: Setting sample rate to 37500
SpyServer: Starting Streaming
gqrx: tpp.c:84: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.
Aborted (core dumped)

uname -a
Linux 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

csete, 2019-04-08:
Like I said in my reply to issue #681, there is nothing we can fix in gqrx. I appreciate that what you see on the screen is gqrx, but please understand that the problem you are having is with an external library that's not even officially supported or tested with gqrx.

closed 2019-04-08 by csete

PA8E, 2019-04-08:
Mea culpa Alexandru, indeed did not understand where to add the bug(still a n00b). 
Will find out where to report it.

csete, 2019-04-08:
Don't worry about, it's very confusing. It's why we need to rewrite everything.