gqrx-sdr/gqrx #684: Raw I/Q passthrough: I and Q data are identical

Vinz87, 2019-04-23:
I thought the "Raw I/Q" option for the demodulator would be a true passthrough of the I/Q complex data, instead I notice that the two channels of the wav file are identical, while I thought I would have found the I data in one channel and the Q data in the other.

Vinz87, 2019-04-25:
Well based on the code I guess it should be like that:
`if (audio_rr)
    {
        // FIXME: DEMOD_NONE has two outputs.
        connect(demod, 0, audio_rr, 0);
    }
    else if (d_demod == NBRX_DEMOD_NONE)
    {
        connect(demod, 0, self(), 0);
        connect(demod, 1, self(), 1);
    }
    else
    {
        connect(demod, 0, self(), 0);
        connect(demod, 0, self(), 1);
    }`

Still, the two channels of the 48 ksps wav file I get are definitely the same, as if the same real o imaginary branch was connected to the audio sink block inputs.

Vinz87, 2019-04-25:
I've gone a little deeper: I can confirm this bug was not present in v2.6-1, where an audio recording with the demodulator set to "Raw I/Q" actually yields a .wav stereo file, in which the two channels are not identical and represent the real and imaginary part of the signal.

Instead, in v2.11.5 and v2.7-1, the same settings produce a .wav stereo file in which the two channels have identical samples.

alexf91, 2019-04-25:
The problem is that narrow band receiver currently contains only one resampler. This was not a problem before the IQ demodulator was added, because SSB and AM have one channel.
When the quadrature rate was increased from 48k to 96k with commit 392dd4206a7e745c8ea047f5312664e8dfeaacbf, resampling to 48k for the audio file became necessary. Since there is only one resampler, both audio channels are connected to the same output. This is what the FIXME comment refers to.
v2.6 had the quadrature rate set to 48k, so the real and imaginary samples were directly connected to the output of the NBRX block, without any resamplers in between.

csete, 2019-04-27:
Thanks for investigating this. It sounds like we can fix it relatively easy, just need a little time.

PR #687 (2019-05-17): Add second resampler for raw IQ demod
Fixes #684 

The second resampler is only used when IQ demod is selected. For all other modes only `audio_rr0` is used like before.
closed 2019-05-17 by csete