gqrx-sdr/gqrx #685: Recording save location and permissions

nardhini, 2019-04-28:
I am using gqrx v2.11.5 and the first time I tried to record an audio file, it didn't have a default location set (the location field was blank) and hence it tired to record the file to the root(/) folder. Since it didn't have permission to write to the root folder, it failed but no error message was displayed. gqrx should have home folder as the default location and throw an error message when permission is not adequate.

csete, 2019-04-28:
Gqrx does use the home directory of the user running the application as default:
https://github.com/csete/gqrx/blob/master/src/qtgui/dockaudio.cpp#L377

closed 2019-04-28 by csete

nardhini, 2019-04-28:
Yes, my bad, the home folder is the default location. But, gqrx doesn't show an error message when the selected folder doesn't have write permission. For example, when I set the location to /root and recorded an audio, it displayed "Recording audio to /root/gqrx*.wav" but the file was not saved at that location.

csete, 2019-04-28:
I agree that there are many behavioral problems with gqrx. Unfortunately, most of these errors occur in external libraries (gnuradio, gr-osmosdr) and it has proven extremely difficult to make proper error reports due to the way these errors are reported. I welcome patches that fix such things.