gqrx-sdr/gqrx #693: Raspberry 4 and Raspbian Buster

sailoog, 2019-07-02:
Trying to test gqrx in the new Raspberry 4 I am getting this error running gqrx-sdr-2.11.5-linux-rpi3:

`error while loading shared libraries: libboost_program_options.so.1.62.0: cannot open shared object file: No such file or directory`

To run the RPi4 you need the latest Raspbian based on Debian Buster. The version of libboost_program_options is higher:

`libboost-program-options-dev is already the newest version (1.67.0.1+b1).
`
Do you have plans to create a version for RPi4?

closed 2019-07-02 by csete

csete, 2019-07-02:
No plans at the moment.

csete, 2019-07-02:
Also, note that Debian Buster already has version 2.11.5 so that should work. If it doesn't report bugs to Debian. It is still in testing so they might have a chance to fix the packages before it becomes stable.

Hosam-M, 2019-07-02:
i have same problem with Pi3B+ and i am using last Version:June 2019 Release date:2019-06-20 Kernel version:4.19 all try show
 error while loading shared libraries: libboost_program_options.so.1.62.0: cannot open shared object file: No such file or directory
or 
No such file or directory 
i dont know if this problem come from 32bit/64bit and bus things as some guys in the web say with another software or not 
so if someone can help we will be thankful

csete, 2019-07-02:
Dear friends,

As the [website](http://gqrx.dk/download/gqrx-sdr-for-the-raspberry-pi) says:

>Gqrx SDR is available as binary package for the Raspberry Pi 3 running Raspbian Stretch and other ARM v6 running a **Debian Stretch** based OS.

And we now know that this binary will not work on **Debian Buster**. However, we also know that the same version is already available in Debian Buster, so please try and use that for now.

Hosam-M, 2019-07-02:
dear i am really thankful for your kinds and patience answering our questions i wish you all the best