gqrx-sdr/gqrx #694: Airspy HF+ no longer recognised after Ubuntu kernel upgrade to HWE kernel.

blacksmock, 2019-08-04:
My installation of Lubuntu 18.04.2 LTS has upgraded to kernel 5.0.0.23 from 4.18.0-25 via Software Update and my Airspy HF+ SDR is no longer recognised. I can get GQRX to recognise it by reverting to the older kernel version. Ask Ubuntu says that this upgrade is correct and that the new kernel is a HWE version (Hardware Enabled version). They recommend reverting to Lubuntu 18.04 or 18.04.1. Has anyone else with this or newer versions of Ubuntu experienced these problems?

dranch, 2019-08-04:
Seems that Ubuntu 18.04 *desktop* versions default to turning on the hardware enablement kernels by default but Ubuntu *server* defaults to the original GA kernels (4.15).  You can configure your desktop version to use the GA kernels (which are still supported and get updates) via here: 

https://wiki.ubuntu.com/Kernel/LTSEnablementStack 
https://askubuntu.com/questions/248914/what-is-hardware-enablement-hwe

csete, 2019-08-12:
Please use the mailing list for technical support: http://gqrx.dk/user-support

closed 2019-08-12 by csete