gqrx-sdr/gqrx #698: [Feature] Enabling CPlotter high quality antialising seems to improve performance of software rasterizer

kantooon, 2019-09-06:
Besides the graphical improvement (which is not very big), setting high quality antialising for QPainter in the plotter seems to improve the performance of the software rasterizer.
Short of using native OpenGL rendering, this seems to make a difference.
The change is minor but I don't have the time to test it in Gqrx. Please see code here:
https://github.com/kantooon/qradiolink/commit/eb0ada48a480a124495888d44699cdbf4c4b513c#diff-ded1df7949ffaf18341ae5f290ef7f40

kantooon, 2019-09-06:
Nevermind, my benchmark was actually wrong.

closed 2019-09-06 by kantooon

csete, 2019-09-07:
Hi @kantooon,

Thanks for the pointer anyway. I think looking into the setRenderHint() options is actually a good idea that I will explore further. Thanks again!

kantooon, 2019-09-07:
You're welcome. I also made some changes to rx_fft_c. Basically the circular buffer was slower to use than a simple vector and memory copy. Also, I think doing the FFT in the work thread of GNU radio made a little more sense. Don't know if these would suit you as I have a different thread architecture.