gqrx-sdr/gqrx #704: rx_fft_c thread uses CPU even when no work is required (enhancement proposal)

kantooon, 2019-09-26:
Right now, rx_fft_c throws samples as they come into the circular buffer even if no FFT is requested. This affects the performance a bit, noticeable especially at high sample rates.
I have modified the code to do the following:

- Don't use a circular buffer and do no work if FFT points are not requested by other threads.
- Skip one memory copy operation by throwing incoming samples directly on the FFT inbuf
- Calculate power spectrum using Volk optimized routines
- Shift the FFT points in rx_fft_c (but not on the GR thread).

This has the effect of lowering the CPU time spent in these calculations (highest effect on higher sample rates). It requires linking directly agains libvolk, but GNU radio already links it. Sorry, don't have the time to do a pull request, but feel free to use the code from here if it suits you: https://github.com/kantooon/qradiolink/blob/next/gr/rx_fft.cpp

csete, 2019-10-27:
Tanks @kantooon, I will take a look at this optimization.

argilo, 2020-11-28:
@kantooon Since you opened this I made a number of performance improvements to the DSP and FFT processing, including using Volk where possible. If there are still further improvements that can be made, feel free to open a pull request or issue.

closed 2020-11-28 by argilo