gqrx-sdr/gqrx #707: Gqrx AppImage does not run

AsciiWolf, 2019-10-22:
The AppImage version of Gqrx crashes with `This application failed to start because it could not find or load the Qt platform plugin "xcb".` error printed on stdout.

This was already reported before as #667, but got closed afterwards. However the issue is still present.

By the way, Gqrx also needs _libnsl.so.1_ that is not bundled in the AppImage. It would be great if the Gqrx AppImage had all the needed libraries included so it would work out-of-box on most big distributions without the need to install any packages.

I am using fully updated Fedora 30 Workstation with default set of packages.

csete, 2019-10-24:
As far as I can tell from a quick google search, libnsl is (used to be) part of glibc, hence there was no need to include it in the AppImage. Perhaps somebody can come up with a good solution.

AsciiWolf, 2020-03-29:
Same issue with the latest 2.12 AppImage on Fedora 31 with Wayland:
1. Before installing _libnsl_: `gqrx-sdr: error while loading shared libraries: libnsl.so.1: cannot open shared object file: No such file or directory`
2. After installing _libnsl_: `This application failed to start because it could not find or load the Qt platform plugin "xcb".`

csete, 2020-03-29:
Where does that AppImage come from? I haven't packaged anything excepot the source code for 2.12...

AsciiWolf, 2020-03-29:
Oops, I thought I downloaded 2.12, but it seems it was 2.11.5. Sorry!

stdevPavelmc, 2020-04-08:
Hi, I have managed to pack GQRX with my own recipe (used on other free software projects...) I have used Ubuntu 16.04 as base and so far I have it tested on 16.04, 18.04 and 20.04 (dev) with no problem.

I can use some feedback from you to polish and make a PR against this repo to automate AppImage generation 

You can find the latest Appimage here under Continuous Builds's Assets.https://github.com/stdevPavelmc/gqrx/releases

stdevPavelmc, 2020-04-08:
You can peek on the code in the travis file here https://github.com/stdevPavelmc/gqrx/blob/appimage/.travis.yml

AsciiWolf, 2020-04-09:
I have just tried your AppImage on my Fedora 31 system and it seems to work fine. Thank you for your great work! I hope these changes will be merged in the upstream Gqrx repo soon. :-)

stdevPavelmc, 2020-04-13:
Nice, that recipe for creating AppImages is tested on other projects ;)

@csete I can create a PR to add the feature of releasing an AppImage on different situations:

1. - Continuous builds
2. - Only in master branch
3. - Only on tagged builds (This is the standard)

The most common choice is 1 & 3, in that way you get a latest (beta) build on every commit to the master branch and a release draft for you to comment and approve on every tagged build (aka every official release)

Take a peek here https://github.com/swl-x/MystiQ/releases to see what it will look. Please comment about this schema and I will put my hands to work ASAP.

csete, 2020-04-14:
Thanks for looking into this @stdevPavelmc.

What we need to close this bug is:
1. A working AppImage of the latest release 2.12.1
2. A script/recipe/procedure to build new AppImages when needed

I am not crazy about the idea of automatic uploads based on git commits or tags for two reasons. A git tag is just a label attached to a particular commit and does not necessarily represent a release or anything else that is meaningful to distribute. The fact that GitHub automatically publishes tags on the release page is quite stupid IMO. Moreover, AppImage is not the primary way of distributing gqrx because it is not complete. In a new installation, using the AppImage requires the user to manually install udev rules and until this drawback is solved AppImage can only be a backup solution.

Not sure what the best way forward is. I guess I can try to build an AppImage using the same procedure that you used and put it in a script that can be used to build AppImages for future releases.

You are welcome to submit a PR that adds continuous build, however, I think I will hold it for future consideration.

stdevPavelmc, 2020-04-15:
Ok, as per your comments I will craft all the magic on a bash script and you can use it as you wish to generate the AppImage when needed.

Thanks.

csete, 2020-04-16:
The script has been merged, however, the issue remains with the AppImage generated on my computer, so I am leaving this issue open for now.

stdevPavelmc, 2020-04-16:
The AppImage files must be generated on as old as possible OS to be compatible with old, actual and future releases.

And also it tends to behave weird on when compiled/crafted on some new/actual OS... at least in my experience, a travis build is running now to generate the tag v2.12.1, I will provide a link to the AppImage file when done.

73/Cheers

csete, 2020-04-18:
AppImage now on the release page, thanks @stdevPavelmc

closed 2020-04-18 by csete