gqrx-sdr/gqrx #708: GQRX Won't open without my device connected

Zeexel, 2019-10-24:
Hello there!

I've spent my night trying to configure an rtl_tcp server on my Raspberry Pi 4, and I heard that GQRX could connect to an rtl_tcp server just fine and listen to it from there, but, unfortunately, whenever I attempt to start GQRX, it refuses unless I have my device connected to it, which ultimately defeats the purpose of the rtl_tcp server.

Here's the log whenever I try to start GQRX on my PC:
`gr-osmosdr v0.1.4-134-gf3905d35 (0.1.5git) gnuradio 3.8.0.0
built-in source types: rtl uhd hackrf bladerf rfspace airspy soapy redpitaya 
[INFO] [UHD] linux; GNU C++ version 9.1.0; Boost_107100; UHD_3.14.1.0-0-unknown
terminate called after throwing an instance of 'std::runtime_error'
  what():  No supported devices found (check the connection and/or udev rules).
Aborted (core dumped)`

kerel-fs, 2019-10-24:
I can reproduce this bug using the exact same versions (archlinux x64 host system).

kerel-fs, 2019-10-24:
Indeed, our gr-osmosdr builds are missing the rtl_tcp support, but this is unrelated to gqrx crashing (which happens due to https://github.com/csete/gqrx/issues/657#issuecomment-423874658 as described earlier).

OT: The missing rtl-tcp support in gr-osmosdr can be fixed by this [patch-00-gr38-compat.txt](https://svnweb.freebsd.org/ports/head/comms/gr-osmosdr/files/patch-01-gr38-blocks-fix.txt?view=markup&sortby=rev&pathrev=509562) from the FreeBSD ports

alexf91, 2019-10-24:
https://github.com/csete/gqrx/issues/657#issuecomment-423874658 causes a crash of Gqrx if no device has been configured before (e.g. on the first start). It then uses /dev/random as its source. I don't think this is the case here because the gr-osmosdr builds without the RTL-TCP source are also missing the file source.

Anyways, this is not an issue of Gqrx besides maybe the crash and the core dump, which is addressed in PR https://github.com/csete/gqrx/pull/686.

205er, 2019-11-13:
Does that mean, that GQRX will only work again, as soon as the osmosdr packages has been updated?
Is there another way to get GQRX running again on my Manjaro installation? 
Sorry if this is a dumb question, but I am not well versed in this subject. ;-)

alexf91, 2019-11-14:
It does work, but only if a supported device is connected.
In the meantime you can build the package yourself with this PKGBUILD:
```
# Maintainer: Kyle Keen <keenerd@gmail.com?>
# Contributor: Dominik Heidler <dheidler@gmail.com>

pkgname=gnuradio-osmosdr
_pkgname=gr-osmosdr
_r=128
_commit=92106ff1275fa57507800811644e454f343c5f6c
pkgver=0.1.5.r$_r.${_commit:0:6}
pkgrel=3
pkgdesc='Source block for Funcube Dongle, RTL-SDR, USRP, OsmoSDR, BladeRF, HackRF and AirSpy devices'
url='https://sdr.osmocom.org/trac/'
arch=('x86_64')
license=('GPL')
depends=('gnuradio' 'gnuradio-iqbal' 'rtl-sdr' 'gnuradio-fcdproplus' 'hackrf' 'bladerf' 'airspy' 'soapysdr')
optdepends=('python-opengl: osmocom_fft')
makedepends=('git' 'cmake' 'boost' 'swig')
replaces=('gr-osmosdr')
#source=("git://git.osmocom.org/gr-osmosdr#commit=$_commit")
# use this random dude's repo, because where is osmocom?
source=("git://github.com/alexf91/gr-osmosdr/#commit=$_commit")
md5sums=('SKIP')

# add support for miri, osmo?

prepare() {
  cd $_pkgname

  #sed -i 's/python$/python2/' apps/osmocom_siggen_base.py
}

build() {
  cd $_pkgname

  mkdir -p build
  cd build

  cmake \
    -Wno-dev \
    -DCMAKE_BUILD_TYPE=Release \
    -DPYTHON_EXECUTABLE="$(which python3)" \
    -DPYTHON_INCLUDE_DIR="$(echo /usr/include/python3*)" \
    -DPYTHON_LIBRARY="$(echo /usr/lib/libpython3.*.so)" \
    -DBoost_NO_BOOST_CMAKE=ON \
    -DCMAKE_INSTALL_PREFIX=/usr ../

  make
}

package() {
  cd $_pkgname/build

  make DESTDIR="$pkgdir" install
}
```
Save it as PKGBUILD in an empty directory and run `makepkg -si` to install it.

Note that this only works with GNU Radio 3.8 and the `gqrx` package in the community repository. `gqrx-git` from the AUR won't work until #705 is merged.

kolbma, 2020-03-04:
Should work now, I think.
Also you'll need to restart to get newly attached devices in config dialog.

alexf91, 2020-10-10:
This can be closed.
The problem has been fixed in the official gr-osmosdr repository and the Arch Linux package is built correctly now.

csete, 2020-10-11:
Thanks for following up :+1:

closed 2020-10-11 by csete