gqrx-sdr/gqrx #710: GQRX Not Starting on Pi 3B

nousernamesavailabel, 2019-11-21:
Like the title says, I have followed all of the install steps and rtl test works, however, I cannot get GQRX to start for the life of me.

reluctantgithubuser, 2019-12-14:
Same issue. It seems to be looking for an older version of libgnuradio than I have.

Running GQRX from command line gives this error:
./gqrx: error while loading shared libraries: libgnuradio-runtime.so.3.7.10: cannot open shared object file: No such file or directory

I have gnuradio 3.7.12 installed.

my system (from uname -a):

Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

Bobby-N7TFQ, 2019-12-14:
ditto to above users. I have followed all steps at least three times to ensure I have checked all the boxes. But when I try to run GQRX from the terminal mode or the GUI interface nothing happens. It is like all my commands go to the bit bucket.
 I have the following info:
gqrx-sdr-2.11.5-linux-rpi3 - installed, currently residing in my Downloads folder
Running Raspberry pi 3B+ and have just this evening completed a complete FULL-UPGRADE prior to the last install.
uname -a responses with:
Linux raspberrypi3b 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

All steps in the GQRX readme file were accomplished in a very methodical fashion.

Please send whatever thoughts my way.

csete, 2019-12-14:
Sorry for the late reply, but I am NOT monitoring this site for user support issues. As the website says, user support is only provided through Gqrx Google Group: http://gqrx.dk/user-support

Please check the archives, there are already many threads regarding installing gqrx on the raspberry pi. It's not trivial but I have seen several messages from people who have got it working even on the latest version of Raspbian.

Note that the binary from the website was built for the previous version of Raspbian OS, Stretch, and will not work on Raspbian Buster. For the latter, you can use the package from the package repository, however, that comes with a set of påroblem of its own. Again, please see the threads on the mailing list.

closed 2019-12-14 by csete