gqrx-sdr/gqrx #711: GQRX received signal SIGSEGV, Segmentation fault.

bahari, 2019-11-25:
Hi,
I'm in the process to debug and develop an application based on gqrx source file as a benchmark using qt5 development ide.

I'm already success compile all the dependencies required for gqrx. I'm also successfully compile gqrx application using qt5 ide without any error.

But the problem is when I run gqrx in the debug mode, the system throw segmentation fault with the message:     Program received signal SIGSEGV, Segmentation fault.

Below are my GDB traces:

GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.3) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gqrx...done.
(gdb) r
Starting program: /home/bahari/MyWorks/sourceCode/gqrx/build-gqrx-Desktop_Qt_5_3_0_GCC_64bit-Debug/gqrx 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff3ca6cd0 in std::locale::locale(std::locale const&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff3ca6cd0 in std::locale::locale(std::locale const&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff2371228 in boost::filesystem::path::imbue(std::locale const&) () from /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.55.0
#2  0x00007ffff23712d1 in boost::filesystem::path::codecvt() () from /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.55.0
#3  0x00007ffff607d9c1 in gr::prefs::_sys_prefs_filenames() () from /usr/local/lib/libgnuradio-runtime-3.7.10.so.0.0.0
#4  0x00007ffff607df54 in gr::prefs::prefs() () from /usr/local/lib/libgnuradio-runtime-3.7.10.so.0.0.0
#5  0x00007ffff607e0d4 in gr::prefs::singleton() () from /usr/local/lib/libgnuradio-runtime-3.7.10.so.0.0.0
#6  0x00007ffff0e48b0a in ?? () from /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so.3.7.10
#7  0x00007ffff7dea1da in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffe3e8, env=env@entry=0x7fffffffe3f8) at dl-init.c:78
#8  0x00007ffff7dea2c3 in call_init (env=<optimized out>, argv=<optimized out>, argc=<optimized out>, l=<optimized out>) at dl-init.c:36
#9  _dl_init (main_map=0x7ffff7ffe1c8, argc=1, argv=0x7fffffffe3e8, env=0x7fffffffe3f8) at dl-init.c:126
#10 0x00007ffff7ddb29a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#11 0x0000000000000001 in ?? ()
#12 0x00007fffffffe686 in ?? ()
#13 0x0000000000000000 in ?? ()
(gdb) 

What is really the cause of this problem? For your information I'm using gnuradio version 3.7.10 and I'm compile it from source. For gr-osmosdr I'm using the latest pull from github and I'm also compile it from source.

kolbma, 2020-03-04:
I think you use some Not-8bit-Charset and there is anywhere a problem with this in the stacktrace. 
Not really related to gqrx, I think.
If you have a Non-ASCII letter in your install path, try to change this and retry.
Or update your system and libraries.

argilo, 2020-11-28:
I don't see Gqrx anywhere in the stack trace, so it looks like the issue is somewhere else. I'll close this off for now, but if you find that it is due to a bug in Gqrx please let me know.

closed 2020-11-28 by argilo