gqrx-sdr/gqrx #712: Feature Request: Waterfall remote control & export

drzraf, 2019-11-25:
A really useful addition to the remote control:
1. Set/Reset waterfall retention time span
2. Export waterfall
3. Reset waterfall.

For such a feature to be *really* nice, some other waterfall-export fixes would be needed:
- image full quality (0.1 sec resolution) even for multi-hours export. For a 1.5 Mhz bandwidth @32k fft, that would be ~1 GB, but it's client responsibility to export & flush before an OOM.
- more precise X graduations (and actually reaching the right border).
- avoid Y time label overlapping on the waterfall.

drzraf, 2024-10-10:
I think https://github.com/gqrx-sdr/gqrx/pull/1289 was a mistake.

Exporting the waterfall is a awesome spectrum analysis feature (like visualizing `rtl_power` without going through an intermediary CSV representation).
With remote control it would be decisive spectrum analysis asset.
(OS screenshot may not be of the same quality nor can it be tightly programmed in relationship to gqrx spectrum analysis)

argilo, 2024-10-10:
I understand that you would find this feature useful, but I don't have the time or energy to build and maintain it. If someone else wants to do so, they are welcome to.