gqrx-sdr/gqrx #715: How to purge gqrx for ubuntu 16.04

patel999jay, 2019-12-14:
My default Linux dpkg messed up after installing gqrx. I tried to reconfigure, it did but when i do update it comes with following error. FYI: I tried removing packages manually, didn't work.

`sudo apt-get autoremove gqrx-sdr
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 libad9361-0 : Depends: libiio0 but it is not going to be installed
 libgnuradio-iio0 : Depends: libiio0 but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
`

csete, 2019-12-14:
Try the commands listed on the website: http://gqrx.dk/download/install-ubuntu

And please use the mailing list for support issues, not this bug tracker:
http://gqrx.dk/user-support

closed 2019-12-14 by csete

patel999jay, 2019-12-14:
Tried all the commands, reconfigure my sources list, but it get failed, 
PS. It not even launch the gqrx itself after installation got successful.