gqrx-sdr/gqrx #717: Problems on macOS 10.15

cxy5014, 2020-01-09:
Since macOS 10.15 banned the 32bit application, Gqrx could not work.
Could you please build the application for macOS 10.15 using 64bit?

csete, 2020-01-09:
Sorry, but I do not have the possibility to build gqrx app bundles for Mac OS X at the moment. Note that gqrx is also available through MacPorts and Homebrew.

kenokenobingo, 2020-01-18:
I can confirm that installing gqrx with Homebrew works.

ra1nb0w, 2020-03-03:
works fine with macports (we are also moving to GNURadio 3.8); you have also gqrx-devel that tracks the latest commit.

argilo, 2020-11-28:
As far as I know, it's possible to build Gqrx on macOS 10.15, since it's working in CI builds.

Binary releases would be nice, so I opened #889 to track that as a feature request.

closed 2020-11-28 by argilo