gqrx-sdr/gqrx #721: Tag a release?

argilo, 2020-01-22:
There have been quite a few useful bug fixes and improvements since 2.11.5 (https://github.com/csete/gqrx/compare/v2.11.5...master), so maybe it would be a good time to tag a release.

argilo, 2020-01-22:
I'll make the necessary code changes and submit a pull request shortly.

csete, 2020-01-25:
Before tagging a release, I need to at least have an idea about how to handle the transition from gnuradio 3.7 to 3.8, and preferably also be able to provide easy to install binary packages and user support. At this time, I am neither physically nor mentally capable of starting that many weeks-long process.

argilo, 2020-01-25:
> I need to at least have an idea about how to handle the transition from gnuradio 3.7 to 3.8

Now that https://github.com/csete/gqrx/pull/705 is merged, Gqrx builds fine with GNURadio 3.7 and 3.8. Is there something more that still needs to be done?

> and preferably also be able to provide easy to install binary packages and user support

How have you built these packages in the past? I could try to help with that.

My motivation for proposing a release tag is that Debian's GNU Radio maintainer is hesitant to update Gqrx without there being a release tag in the git repository. I would really like to see the recent improvements make it into Debian within the next few weeks so they'll end up in Ubuntu 20.04.

csete, 2020-01-25:
> I would really like to see the recent improvements make it into Debian within the next few weeks so they'll end up in Ubuntu 20.04.

Good point! This alone is a very good reason indeed.

> Now that #705 is merged, Gqrx builds fine with GNURadio 3.7 and 3.8. Is there something more that still needs to be done?

I'm concerned about gr-osmosdr. AFAIK it doesn't have official support for GNU Radio 3.8, so I'm wondering how that will fit in the next release of Debian, Ubuntu, etc. Do you have any information about it?

> How have you built these packages in the past? I could try to help with that.

I maintained both an Ubuntu PPA, AppImage and even a tar.gz with binaries for Linux and MacOS app bundle. These included a recent version of GNU Radio, driver libraries and patched gr-osmosdr, so it was a huge undertaking. However, if gqrx 2.6 can make it into Ubuntu 20.04 and it works, the PPA will no longer be necessary and that would be a huge help.

So, if you think there is a chance that 2.6 can make it into Ubuntu 20.04 in a "working state", I will agree to make the tag now, and worry about binaries later :)

csete, 2020-01-25:
I can see that Debian Testing already uses patches that builds gqrx and gr-osmosdr with GNU Radio 3.8. I am going to test it.

argilo, 2020-01-25:
That's correct -- Debian unstable (and testing) were already using forks of gqrx and gr-osmosdr that added GNU Radio 3.8 support. The gr-osmosdr project now has an official `gr3.8` branch, so I presume Debian will switch to that. I'll check in with Maitland to see what the plan is.

devnulling, 2020-01-25:
I've tested the [gr3.8](https://github.com/osmocom/gr-osmosdr/tree/gr3.8) branch of gr-osmosdr on Ubuntu 18.04.x with gqrx. The one caveat people may run into (on 18.x or other OSes) is that they will need to update their cmake to be >3.13.

argilo, 2020-01-25:
By the way, I just tested the gqrx-sdr package on the latest daily build of Ubuntu 20.04 and it's running fine with GNU Radio 3.8. All that's missing are the recent Gqrx bug fixes, and those should flow into Debian unstable/testing and Ubuntu 20.04 as soon as a tag is added here.

argilo, 2020-01-25:
> I maintained both an Ubuntu PPA, AppImage and even a tar.gz with binaries for Linux and MacOS app bundle. These included a recent version of GNU Radio, driver libraries and patched gr-osmosdr, so it was a huge undertaking. However, if gqrx 2.12 can make it into Ubuntu 20.04 and it works, the PPA will no longer be necessary and that would be a huge help.

I used to use your PPA, and it was very helpful before all the various SDR packages made their way into Debian & Ubuntu. Now that things have matured and stabilized, I agree that the PPA is no longer needed. For anyone who wants to be on the bleeding edge there's PyBOMBS.

closed 2020-01-25 by csete

argilo, 2020-01-26:
Note: this was auto-closed by #722 but it's not quite resolved yet since version 2.12 still needs a git tag.

csete, 2020-01-26:
> Note: this was auto-closed by #722 but it's not quite resolved yet since version 2.12 still needs a git tag.

Yes, I just wanted to apply some more patches and test a little before tagging a release. I'm also considering to include PR #635 from @alexf91 before tagging but I will create the tag by the end of the day.

csete, 2020-01-26:
> I've tested the [gr3.8](https://github.com/osmocom/gr-osmosdr/tree/gr3.8) branch of gr-osmosdr on Ubuntu 18.04.x with gqrx. The one caveat people may run into (on 18.x or other OSes) is that they will need to update their cmake to be >3.13.

I tried it too on my Ubuntu 18.04 but couldn't build. Probably because of the cmake version, though it didn't complain, it just didn't generate any Makefile. I got it working by taking the master branch and applying the patches from Debian manually.

The daily Ubuntu 20.04 works fine for me too, so I'm good with going ahead.

argilo, 2020-01-26:
> I tried it too on my Ubuntu 18.04 but couldn't build. Probably because of the cmake version

Yes, it's because the `gr3.8` branch now requires cmake 3.13. I got around that by removing the cmake package and installing the latest cmake from source.

argilo, 2020-01-26:
Before I did that, cmake did complain that the required version was too high though. I'm surprised you didn't get an error. :confused:

argilo, 2020-01-26:
> Yes, I just wanted to apply some more patches and test a little before tagging a release. I'm also considering to include PR #635 from @alexf91 before tagging but I will create the tag by the end of the day.

Perfect. That will leave plenty of time before Ubuntu 20.04's Debian import freeze. I'll be happy to test out any further changes as well.

csete, 2020-01-26:
Tag created :)

amontefusco, 2020-01-26:
Compiles (both cmake and qmake) and runs successfully on U18.04 with GnuRadio 3.7.11, gnuradio-osmosdr compiled locally from my fork.

PR #722 (2020-01-25): Version 2.12
Fixes #721.

Making a release will let the recent improvements trickle out to users of distributions like Debian. I've numbered this release 2.12 since it has some new things (GNU Radio 3.8 support, audio mute button).

The following changes are included: https://github.com/csete/gqrx/compare/v2.11.5...34333e78b2e565745b6d62bf0efcc57cc3b1988f

Once this is merged a `v2.12` tag will need to be pushed.