gqrx-sdr/gqrx #723: Core dumped on playing IO record

0MazaHacka0, 2020-01-25:
Manjaro 18.1 x64, gqrx 2.11.5

1. Open GQRX
2. Record IO
3. Try to play recorded IO -> Crash
4. Stop playing from sdr, try to play from io -> crash

qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
gr-osmosdr v0.1.4-134-gf3905d35 (0.1.5git) gnuradio 3.8.0.0
built-in source types: rtl uhd hackrf bladerf rfspace airspy soapy redpitaya 
[INFO] [UHD] linux; GNU C++ version 9.2.0; Boost_107100; UHD_3.15.0.0-0-unknown
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Resampling audio 96000 -> 48000
BookmarksFile is /home/atom/.config/gqrx/bookmarks.csv
gr-osmosdr v0.1.4-134-gf3905d35 (0.1.5git) gnuradio 3.8.0.0
built-in source types: rtl uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
gr-osmosdr v0.1.4-134-gf3905d35 (0.1.5git) gnuradio 3.8.0.0
built-in source types: rtl uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
usb_claim_interface error -6
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to open rtlsdr device.
Aborted (core dumped)

alexf91, 2020-01-25:
Are you trying to replay in a different instance of Gqrx while the old one is still running?

0MazaHacka0, 2020-01-25:
No, only one instance is running

lukasz-zaroda, 2020-01-26:
I have exactly the same problem.

`Arch Linux 5.4.14-arch1-1`
`gqrx 2.11.5`

csete, 2020-01-26:
I can't reproduce it locally, but as the output indicates, something is blocking access to the RTL-SDR device, which is what causes the crash. Note, however, that there are no "playack" messages in the output and I have difficulties making that output fit with the sequence of events you describe.

0MazaHacka0, 2020-01-26:
All I can say is that older version worked perfectly. I'll try to find which one worked

alexf91, 2020-01-27:
I can reproduce it with the gnuradio-osmosdr package from the Arch community repository.
This package has several problems. I've created a bug report a few months ago related to the rtl-tcp and file backends, but there has not been any response or fix. #708 addresses some of the issues.
You can build the package yourself with the PKGBUILD in this comment: https://github.com/csete/gqrx/issues/708#issuecomment-553723163
There might still be other issues, but at least Gqrx doesn't crash.

alexf91, 2020-10-10:
This can be closed.
The problem was with the Arch Linux package, which has been fixed.

csete, 2020-10-11:
Thanks for following up :+1:

closed 2020-10-11 by csete